eProsima/Micro-XRCE-DDS-Client #4: When subscriber use best effort stream, an error occurs.

muto-esol, 2018-06-08:
I'm trying to HelloWorld examples which connect to FastRTPS.
I confirmed client can subscribe messages using reliable stream.
But when using best-effort stream, client is not working as desired.

I get this error:

```
UDP mode => ip: 127.0.0.1 - port: 2020 >>
Receive topic: HelloWorld, count: 1
Connection error at read_data
```

### Environment

OS: Ubuntu 16.04
Following examples execute in PC:

```
publisher: FastRTPS example
HelloWorldExample  
      |  
micro RTPS agent  
      |  
subscriber: micro RTPS client example  
SubscribeHelloWorld  
```

And I changed the client's code to use the best-effort stream.
Below is my code:

```c
examples/SubscribeHelloWorld/main.c

...(snipped)...
    /* Main loop */
    while(true)
    {
        /* Request data */
        read_data_sync(&my_session, datareader_id, STREAMID_BUILTIN_BEST_EFFORTS); // Changed here
        check_and_print_error(&my_session, "read_data");

        run_communication(&my_session);

        ms_sleep(100);
    }
...(snipped)...
```

### Results

FastRTPS Puslisher send a lot of messages.
such as:

```
Starting 
Publisher matched
Message: HelloWorld with index: 1 SENT
Message: HelloWorld with index: 2 SENT
Message: HelloWorld with index: 3 SENT
Message: HelloWorld with index: 4 SENT
Message: HelloWorld with index: 5 SENT
Message: HelloWorld with index: 6 SENT
Message: HelloWorld with index: 7 SENT
Message: HelloWorld with index: 8 SENT
Message: HelloWorld with index: 9 SENT
Message: HelloWorld with index: 10 SENT
```

But client just receive one message and error occurs.
such as:

```
<< UDP mode => ip: 127.0.0.1 - port: 2020 >>
Receive topic: HelloWorld, count: 1
Connection error at read_data
```

What could be the cause of this error?

Appreciate your help.

lemunozm, 2018-06-12:
Hi @muto-esol, thank you for your report. We have released a new version that fix this issue. The changes only affect to the agent side.

muto-esol, 2018-06-19:
It's working as expected. Thanks for the quick fix.

closed 2018-06-19 by richiware