eProsima/Micro-XRCE-DDS-Client #5: Question: direction of inequility sign is right?

muto-esol, 2018-06-29:
I think direction of inequility sign is inverse. 

https://github.com/eProsima/micro-RTPS-client/blob/40f9335fe85739299842c4a80045510b08620010/examples/PublishHelloWorld/HelloWorld.h#L79-L86

"stream_id" of Best effort is 0x01 and  "stream_id" of Reliable is 0x80.
So I think direction of inequality sign should be like this:

```
79    if (128 > stream_id)
```

Is this the right?

Appreciate your help.

richiware, 2018-06-29:
Commit a1299aa2b148bdee0b2c7159309c572c374d09e8 solves it. Commit 2bdfc7d968c3a6dcb85ab807757390d67f285e6a solves the 
problem in the code generator. Thanks for the report.

muto-esol, 2018-07-02:
"Shape.h" is also need to be fixed?
https://github.com/eProsima/micro-RTPS-client/blob/a1299aa2b148bdee0b2c7159309c572c374d09e8/examples/ShapesDemo/Shape.h#L87-L94

richiware, 2018-07-02:
Thanks for the report. Commit 22c196baf5c1a8629c151ab608e2d065c21642cc updates also `Shapes.h`.

muto-esol, 2018-07-02:
Thanks @richiware  I confirmed that it fixes the issue for us.

closed 2018-07-02 by muto-esol