eProsima/Micro-XRCE-DDS-Client #6: long run stability issues

muto-esol, 2018-06-29:
I'm trying to HelloWorld example which connect to agent by Serial.
Client is subscriber and use reliable stream.
It seems to be working fine but when running for a long time(over an hour), I got the following error.

```
Connection error at read_data
```

## Environment
OS: Ubuntu 16.04
Following examples execute in PC(client and agent  are executed in the same PC):

```
publisher: FastRTPS example
HelloWorldExample
      |  
micro RTPS agent
      | USB port: /dev/ttyUSB0
      |
      | Serial-to-USB Cable
      |
      | Serial port:/dev/ttyS0
subscriber: micro RTPS client example
SubscribeHelloWorld
```

And I changed the FastRTPS HelloWorldExample.
Below is my code:

Fast-RTPS/examples/C++/HelloWorldExample/HelloWorldPublisher.cpp

```
108         eClock::my_sleep(1000);   //  Modified: Publish at every 1 second
```

Fast-RTPS/examples/C++/HelloWorldExample/HelloWorld_main.cpp

```
57          mypub.run(10000);   // Modified: Run for a long time
```

## Results

After about an hour,  I got the following error.

such as:

```
...(snipped)...
Receive topic: HelloWorld, count: 4322
Connection error at read_data
```

What could be the cause of this problem?

Appreciate your help.

lemunozm, 2018-08-27:
Hi @muto-esol, in the new version we have tested the example by several hours without connection problems. If you test it this version and continue with the same problem, please, tell us.

muto-esol, 2018-08-29:
@lemunozm Thanks for confirming!
I will reopen this issue if it reoccurs.

closed 2018-08-29 by muto-esol