eProsima/Micro-XRCE-DDS-Client #7: Is it possible to publish/subscribe large data?

muto-esol, 2018-07-04:
I'm trying to HelloWorld examples which connect to FastRTPS and publish/subscribe large data.
But client cannot publish/subscribe data.

## Environment
OS: Ubuntu 16.04
Following examples execute in PC:

```
subscriber: FastRTPS example
HelloWorldExample  
      |  
micro RTPS agent  
      |  
publisher: micro RTPS client example  
PublishHelloWorld  
```

And I changed the micro-RTPS-client example.
Below is my code:

micro-RTPS-client/examples/PublishHelloWorld/main.c

```
114        topic.message = "AAAAA.................";　// over 500 characters
```

## Results
I captured network packets (wireshark) but client didn't send messages.

What am I doing wrong? configuration is wrong or memory problem?
Thanks for the help.

julianbermudez, 2018-07-04:
Hi @muto-esol, 
There is a MTU size (512 bytes by default) which limits the maximum message size. In order to increase this MTU, you have to modified the `CONFIG_MAX_TRANSMISSION_UNIT_SIZE` define in `transport/config.h` for both Agent and Client. As well as modified the `MICRORTPS_MTU_SIZE` define in `xrce_client.h`. Please let's me known if it works for you.

julianbermudez, 2018-08-07:
Hi @muto-esol, 
We have just launched a new release 1.0.0beta2 in which we are improved the MTU selection. I close the issue, in case of persistence reopen the issue.

closed 2018-08-07 by julianbermudez

ToRaVe, 2018-10-05:
Hello there,
I tried to increase the MTU for UDP by setting
- on the client side inside client.config: CONFIG_UDP_TRANSPORT_MTU=1024
- on the agent side inside CMakeLists.txt: CONFIG_UDP_TRANSPORT_MTU=1024
Then I built both and tried to run the agent and two clients, but I always get
`serialize eprosima::fastcdr::exception::NotEnoughMemoryException`

am I missing something?

julianbermudez, 2018-10-08:
Hi @ToRaVe,

Your error is due to the MTU of the output packets from the Agent to the Client, you can solve it change the `mtu_size` in `OutputMessage.h`. As you can see there is a related TODO which we are going to solve for next release which will be available in the next weeks. I hope it helps you.

ToRaVe, 2018-10-08:
Good morning @julianbermudez ,

thank you very much, this solved the problem!
However, when I increase the mtu size, the agent gives me a segmentation fault. A value of 2048 works perfectly fine, but anything above 3000 seems to be problematic. I'm not sure where this error originates -does it have to do with an underlying layer or may my system be the problem?

julianbermudez, 2018-10-08:
Hi @ToRaVe,

I will check it in order to give you an answer as soon as possible.

julianbermudez, 2018-10-23:
Hi @ToRaVe,

I checked and solved your issue on commit https://github.com/eProsima/Micro-XRCE-DDS-Agent/commit/bd3967cd380f2e59c4fa5cd64f88c4e4d150d463.
Apart from this hotfix you should configure publisher and subscriber as is described in https://github.com/eProsima/Micro-XRCE-DDS-Agent/issues/4 in order to send large data. This issue could be relate with #24. 
Please check it and give me your feedback.
Thanks!

ToRaVe, 2018-11-05:
Hello @julianbermudez,

Sorry for my late answer, I've been sick. 
I wanted to test the software, but I haven't actually been able to build. I cloned the master repository of the Micro-XRCE-DDS-Agent and followed the instructions on https://micro-xrce-dds.readthedocs.io/en/latest/installation.html. When I use the make command, I get the following output:
```
Scanning dependencies of target microxrcedds_agent
[  3%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/Root.cpp.o
[  6%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/processor/Processor.cpp.o
[ 10%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/client/ProxyClient.cpp.o
[ 13%] Building CXX object CMakeFiles/microxrcedds_agent.dir/src/cpp/participant/Participant.cpp.o
/home/tv/Projects/Micro-XRCE-DDS-Agent/src/cpp/participant/Participant.cpp: In member function ‘virtual bool eprosima::uxr::Participant::matched(const dds::xrce::ObjectVariant&) const’:
/home/tv/Projects/Micro-XRCE-DDS-Agent/src/cpp/participant/Participant.cpp:152:43: error: no match for ‘operator==’ (operand types are ‘eprosima::fastrtps::ParticipantAttributes’ and ‘const eprosima::fastrtps::ParticipantAttributes’)
     return parser_cond && (new_attributes == old_attributes);
                            ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~
CMakeFiles/microxrcedds_agent.dir/build.make:134: recipe for target 'CMakeFiles/microxrcedds_agent.dir/src/cpp/participant/Participant.cpp.o' failed
make[2]: *** [CMakeFiles/microxrcedds_agent.dir/src/cpp/participant/Participant.cpp.o] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/microxrcedds_agent.dir/all' failed
make[1]: *** [CMakeFiles/microxrcedds_agent.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```
Should I open a new issue for this, since it isn't really related to the datasize affair?

julianbermudez, 2018-11-05:
Hi @ToRaVe, 

There is a similar issue open, I redirect to [it](https://github.com/eProsima/Micro-XRCE-DDS/issues/14). Please tell me if you are able to solve this issue. Thanks!