eProsima/Micro-XRCE-DDS-Client #8: Delete a specific entity resource.

OpusK, 2018-08-29:
Hi,

The client has an API to remove the session.
However, there is no API to delete each Entity.
I want to delete Entity only while maintaining Session.

Do you plan to add these APIs?

lemunozm, 2018-09-05:
Hi @OpusK,

I think the function you are looking for is the following:
```c
uint16_t mr_write_delete_entity(mrSession* session, mrStreamId stream_id, mrObjectId object_id);
```
It is placed in `include/micrortps/client/profile/session/common_create_entities.h` and is included automatically when you include `client.h` if you have enable the `PROFILE_CREATE_ENTITIES_REF` or `PROFILE_CREATE_ENTITIES_XML` into `client.config` file.

OpusK, 2018-09-06:
Hi, @lemunozm. 

Thank you! It works well as I want!

closed 2018-09-06 by OpusK