eProsima/Micro-XRCE-DDS-Client #9: create participant using xml string.

OpusK, 2018-09-03:
Hi,

In the current version I know that I can only create participant through mr_write_create_participant_ref().
Will you also support REPRESENTATION_AS_XML_STRING method?
And Agent only supports "default participant" for REPRESENTATION_BY_REFERENCE method.
Will you support other strings later?

julianbermudez, 2018-09-03:
Hi @OpusK, 

Yes there is a `mr_configure_participant_xml` function which allows you create participant by XML.
It is also planning to create a customizable database for REF entities. It is expected that we launch this feature in the coming weeks.

OpusK, 2018-09-06:
Hi, @julianbermudez,

I look forward to the update. Thank you.

OpusK, 2018-10-11:
@julianbermudez,

Cool! Thanks for your hard works!

closed 2018-10-11 by OpusK