eProsima/Micro-XRCE-DDS-Client #52: [1.0.1 Version] Related to XML and entity deletion.

OpusK, 2018-11-19:
Hi,

With the new release, the wiki introduces the following example to use XML.

```c
uxrObjectId datareader_id = uxr_object_id(0x01, UXR_DATAREADER_ID);
const char* datareader_xml = "<dds>"
                                     "<data_reader>"
                                         "<topic>"
                                             "<kind>NO_KEY</kind>"
                                             "<name>HelloWorldTopic</name>"
                                             "<dataType>HelloWorld</dataType>"
                                         "</topic>"
                                     "</data_reader>"
                                 "</dds>";
uint16_t datareader_req = uxr_buffer_create_datareader_xml(&session, reliable_out, datareader_id, subscriber_id, datareader_xml, UXR_REPLACE);

```

In this case, however, neither the reader nor the writer (nor the participant) is created.
Rather, it works with older versions of XML styles like below. Is there something I'm using wrong?
```
"<profiles><subscriber profile_name=\"default_xrce_subscriber_profile\"><topic><kind>NO_KEY</kind><name>HelloWorldTopic</name><dataType>HelloWorld</dataType><historyQos><kind>KEEP_LAST</kind><depth>1</depth></historyQos><durability><kind>TRANSIENT_LOCAL</kind></durability></topic></subscriber></profiles>"
```

Next, there is issue related to entitiy deletion.
1.0.0 Beta2 is a feature I used well (used by XEL Network).
However, it does not work in newer versions. The code is used in the following manner. Has the usage changed?
```cpp
uxrObjectId obj_id = {sub_list_[i]->reader_id_, UXR_DATAREADER_ID};
uxr_buffer_delete_entity(participant_.session, participant_.output_stream_id, obj_id);
```

lemunozm, 2018-11-20:
ROS2 comes with FastRTPS 1.6.0. The agent links dynamically with the FastRTPS. It is possible that when you run the Agent with the ROS2 installed, the Agent is linking with the 1.6.0 (although it was compiled with the headers of 1.7.0, because you used the THIRDPARTY installation). Usually, it would cause a segmentation fault, but if the interface does not change a lot, it would appear that all things go well.

As a temporal workaround, I recommend you to link the agent statically with the FastRTPS 1.7.0, using the CMake flag: `-DCMAKE_SHARED_LIBS=OFF`

I hope this fix your problem temporally until ROS2 uses the FastRTPS 1.7.0.

OpusK, 2018-11-20:
Thanks @lemunozm,

But, the flag(CMAKE_SHARED_LIBS) you gave is not in CMAKELIST. Did you change to another name?

lemunozm, 2018-11-20:
It is `-DBUILD_SHARED_LIBS=OFF`, I have corrected in the previous message.
My apologies!

OpusK, 2018-11-21:
I installed it using the flag you gave me. However, if I activate the configuration of ROS2, the link address is changed as follows (for ROS2)

```bash
$ ldd libmicroxrcedds_agent.so
	linux-vdso.so.1 (0x00007ffeaa510000)
	libfastrtps.so.1 => /opt/ros/bouncy/lib/libfastrtps.so.1 (0x00007f9f7704f000)
	libfastcdr.so.1 => /usr/local/lib/libfastcdr.so.1 (0x00007f9f76e3c000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f9f76c1d000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f9f76894000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f9f7667c000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f9f7628b000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f9f77845000)
	libfastcdr.so.1.0.7 => /opt/ros/bouncy/lib/libfastcdr.so.1.0.7 (0x00007f9f7607b000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f9f75e77000)
	libcrypto.so.1.1 => /usr/lib/x86_64-linux-gnu/libcrypto.so.1.1 (0x00007f9f759ff000)
	libtinyxml2.so.6 => /usr/lib/x86_64-linux-gnu/libtinyxml2.so.6 (0x00007f9f757eb000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f9f7544d000)
```

Currently, I am using ROS2 as a binary package, so I'll change to source build to apply the second option you gave me.

OpusK, 2018-11-21:
@lemunozm ,

It seems that rmw_fastrtps does not have a branch that supports FastRTPS 1.7.0.
So, the second option seems to be unable to try ...

```bash
--- stderr: rmw_fastrtps_shared_cpp                                                                                  
In file included from /home/kei/workspace/ros2/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_client.cpp:25:0:
/home/kei/workspace/ros2/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/custom_participant_info.hpp:45:8: error: ‘void ParticipantListener::onParticipantDiscovery(eprosima::fastrtps::Participant*, eprosima::fastrtps::rtps::ParticipantDiscoveryInfo)’ marked ‘override’, but does not override
   void onParticipantDiscovery(
        ^~~~~~~~~~~~~~~~~~~~~~
/home/kei/workspace/ros2/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/custom_participant_info.hpp: In member function ‘void ParticipantListener::onParticipantDiscovery(eprosima::fastrtps::Participant*, eprosima::fastrtps::rtps::ParticipantDiscoveryInfo)’:
/home/kei/workspace/ros2/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/custom_participant_info.hpp:50:12: error: ‘struct eprosima::fastrtps::rtps::ParticipantDiscoveryInfo’ has no member named ‘rtps’
       info.rtps.m_status != eprosima::fastrtps::rtps::DISCOVERED_RTPSPARTICIPANT &&
            ^~~~
/home/kei/workspace/ros2/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/custom_participant_info.hpp:50:55: error: ‘DISCOVERED_RTPSPARTICIPANT’ is not a member of ‘eprosima::fastrtps::rtps’
       info.rtps.m_status != eprosima::fastrtps::rtps::DISCOVERED_RTPSPARTICIPANT &&
                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/kei/workspace/ros2/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/custom_participant_info.hpp:51:12: error: ‘struct eprosima::fastrtps::rtps::ParticipantDiscoveryInfo’ has no member named ‘rtps’
       info.rtps.m_status != eprosima::fastrtps::rtps::REMOVED_RTPSPARTICIPANT &&
            ^~~~
/home/kei/workspace/ros2/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/custom_participant_info.hpp:51:55: error: ‘REMOVED_RTPSPARTICIPANT’ is not a member of ‘eprosima::fastrtps::rtps’
       info.rtps.m_status != eprosima::fastrtps::rtps::REMOVED_RTPSPARTICIPANT &&
                                                       ^~~~~~~~~~~~~~~~~~~~~~~
..
..
..
<more--->

```

lemunozm, 2018-11-21:
Hi @OpusK ,

Referred to the static compile flag, I have tested it in my computer, and the link to libfastrtps.so.1 does not appear, as expected. Try to clean the build folder and run the following:

```
cmake -DTHIRDPARTY=ON -DBUILD_SHARED_LIBS=OFF -DCMAKE_POSITION_INDEPENDENT_CODE=ON ..
make
```
(The `-DCMAKE_POSITION_INDEPENDENT_CODE=ON` add the `-fPIC` necessary when compile statically)
After it, the `ldd MicroXRCEAgent` will not show the link to FastRTPS.

lemunozm, 2018-11-22:
Hi @OpusK,

Do not worry about the version tag of `config.h` file. The agent in version 1.0.1 is pointing to a FastRTPS commit with the features of 1.7 but without the version tags changed. If you compile statically, the FastRTPS that the agent uses is the correct one.

OpusK, 2018-11-23:
Hi, @lemunozm 

Okay. I will test it further. 
Thanks for your support :)

OpusK, 2018-12-07:
These issues have been resolved with rmw_fastrtps updated to 1.7.0 and XRCE-DDS-Agent updated.
Thank you @lemunozm.

closed 2018-12-07 by OpusK