eProsima/Micro-XRCE-DDS-Client #61: Allow user defined settings

OpusK, 2018-12-19:
Hi,

Currently, there is the following issue.
 - [allow user defined settings](https://github.com/ROBOTIS-GIT/ros2arduino/issues/1)

In particular, the definition mentioned in the link above is maintained in internal.h.

Users can find and fix these things one by one. 
However, I thought that it would be nice if the user could define it all at once by a configuration file (eg. config.h)

What do you think about this?

lemunozm, 2018-12-19:
Hi @OpusK,

These definitions are internal of the DDS-XRCE management protocol. If some of these definitions correspond to buffers size, is because the protocol needs this size to work properly, and must not be modified. 

Also, there are some parts of the protocol that are not currently added, for example, the `UXR_PACKED_SAMPLES_SEQUENCE_MAX` defined in `xrce_protocol_internal.h` file is used to define a buffer size that will never be placed in the user stack. In the future, if this value affects to the memory stack size and can be modifiable by the user, it will be accessible from the `client.config` file.

The only definition of `xrce_protocol_internal.h` file that is using a valid buffer in the stack is `UXR_STRING_SIZE_MAX`, but this buffer is only created if the log is enabled (compiled with the CMake option -DVERBOSE_MESSAGE=ON) used for debug purposes.

OpusK, 2018-12-20:
Hi, @lemunozm 

Thank you for the detailed and friendly explanation.
I fully understood your words :) Thank you!

closed 2018-12-20 by OpusK