eProsima/Micro-XRCE-DDS-Client #72: Is there some examples about effort pub/sub mode?

baozeng90, 2019-01-24:
Is there some examples about effort pub/sub mode?

lemunozm, 2019-01-25:
Hi @baozeng90, 

The external API behaviour is the same for both, best effort and reliable modes. 
The only part you need to modify is the initialization. For example, in the case of  publishing, you will initialize the output reliable stream as follows:
```c
uint8_t buffer[BUFFER_SIZE];                                                                 
uxrStreamId id = uxr_create_output_reliable_stream(&session, buffer, BUFFER_SIZE, STREAM_HISTORY);
```
For best effort mode it should be:
```c
uint8_t buffer[BUFFER_SIZE];                                                                 
uxrStreamId id = uxr_create_output_best_effort_stream(&session, buffer, BUFFER_SIZE);
```
The doc about this can be found at https://micro-xrce-dds.readthedocs.io/en/latest/client.html#session
Also, you have the `ShapesDemo` example with both, best effort and reliable communication.

Regards.

baozeng90, 2019-01-25:
@lemunozm 
1）uxr_init_session_info(&session->info,0x81, key); session id 0x81 need modify ?

2) if(!uxr_run_session_until_all_status(&session, 1000, requests, status, 4))/ it alse use uxr_run_session_until_all_status to send create entities message and wait its status?

closed 2019-02-05 by lemunozm

eranroll, 2021-01-23:
Hi,

I know this is quite an old thread, but are there any more elaborate examples?
I can't seem to get best_effort to work on an STM embedded device using Micro-ROS.

Regards