eProsima/Micro-XRCE-DDS-Client #79: Large size topic transmission problem.

OpusK, 2019-02-22:
Hi

We use Client and Agent in RaspberryPi to transfer the LIDAR data in TB3.
- [Turtlebot3 LIDAR client](https://github.com/ROBOTIS-GIT/turtlebot3/blob/ros2/turtlebot3_lidar/turtlebot3_lidar/main.cpp)

We have been using the LIDAR data of 360 degrees to 180 degrees so far as it is urgent.
But now I want to transfer all the data from the Client.

As a result of my testing, up to 240 data can be transferred without any problem. However, starting from 241, the following error occurs in Agent.
```bash
Error written data on DDS entity.
```

I did not calculate the exact size of the data, but when I send 240 data, it contains about 1000 bytes if I include the remaining data.

Let me know if you need more information.

julianbermudez, 2019-02-22:
Hi @OpusK, 

Did you take into account the explanation provided in [this issue](https://github.com/eProsima/Micro-XRCE-DDS-Agent/issues/4)?

julianbermudez, 2019-04-08:
Hi @OpusK,

I close the issue. If you continue with the problem, please reopen it.

closed 2019-04-08 by julianbermudez

OpusK, 2019-05-14:
Hi, @julianbermudez 

This method does not work.
There may be a problem with my XML expression. So, can you give an example of how to write XML for writers and readers on the client? (including details about datawriter)