eProsima/Micro-XRCE-DDS-Client #80: About UXR_REUSE flag.

OpusK, 2019-03-07:
Hi,

I am using UXR_REUSE or UXR_REPLACE flags when creating entities.

When creating the Topic, we use the following function parameters.
- [git source link](https://github.com/ROBOTIS-GIT/ros2xrcedds/blob/53e6bdb7c9537bd23693bf1d3316f2c760cd65d3/xrcedds/xrcedds.cpp#L162)

However, UXR_REUSE, as I expected, can not use the existing Topic Type, and the Agent is returning only the failure message as below.
```bash
2019-03-07 13:47:48.323 [PARTICIPANT Error] Type with the same name already exists:sensor_msgs::msg::dds_::BatteryState_ -> Function registerType
2019-03-07 13:47:48.830 [PARTICIPANT Error] Type with the same name already exists:sensor_msgs::msg::dds_::Joy_ -> Function registerType
```

Is not this flag implemented yet? Or is there something I'm using wrong?

julianbermudez, 2019-03-07:
Hi @OpusK,

The problem is that you are using the `UXR_REUSE` flag over different XRCE-Objects (different `topic_ids`) and this flag only makes sense when you want to reuse the same XRCE-Object (same `topic_id`).
The error that the Agent shows comes from Fast-RTPS, you cannot register the same RTPS-Topic in the same RTPS-Participant more than once.

The following schema summarizes the error:

```
XRCE-Topic (id:1, xml:"topic") ---> XRCE-Participant (id:1) ---> RTPS-Participant (id:1) ---> RTPS-Topic (name: "topic") ---> OK.
XRCE-Topic (id:2, xml:"topic") ---> XRCE-Participant (id:1) ---> RTPS-Participant (id:1) ---> RTPS-Topic (name: "topic") ---> ERROR: already exits.
```

OpusK, 2019-03-11:
Hi, @julianbermudez 

Thanks for your support!

closed 2019-03-11 by OpusK