eProsima/Micro-XRCE-DDS-Client #83: Hello DDS world subscriber always stop at 65535

kelvin-vincent, 2019-03-29:
Hi dear,
I had setup the development environment according to MicroXRCE-DDS Documentation Release 1.0.0. I setup it in the Ubuntu 18.04 and compile the examples. 

I ran the agent first, then ran the example programs, that is, the HellowWorld publisher and subscriber examples, in the same PC running Ubuntu.

However, after a day of running, subscriber will always stop with id = 65535. It seemed not publisher's problem since the publisher continues to send topic... and if I stop and restart the subscriber, the receiving will continue. I had verified this situation several times. Do you see similar situation? Please advise what possible is the problem. Thanks.

kelvin-vincent, 2019-04-02:
Hello,
Can any help me? Thanks.

lemunozm, 2019-04-02:
Hi @kelvin-vincent, I will test it ASAP and I will answer you. 
What version are you using? Could you tell me the commits of both, client and agent?
Thanks.

lemunozm, 2019-04-02:
Hi @kelvin-vincent, in this PR you have the issue solved: https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/68
Soon it will be merged to master branch.
Thank for reporting the problem.

lemunozm, 2019-04-04:
Hi @kelvin-vincent, the https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/68 PR has been merged to master.

closed 2019-04-12 by lemunozm