eProsima/Micro-XRCE-DDS-Client #89: deserialize eprosima::fastcdr::exception::NotEnoughMemoryException

OpusK, 2019-05-16:
Hi, @julianbermudez 

```
<historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
```

This method does not work.
There may be a problem with my XML expression. So, can you give an example of how to write XML for writers and readers on the client? (including details about datawriter)

_Originally posted by @OpusK in https://github.com/eProsima/Micro-XRCE-DDS-Client/issues/79#issuecomment-492139442_

Below is the error message.

```
deserialize eprosima::fastcdr::exception::NotEnoughMemoryException
Error processing WRITE_DATA submessage.
Error written data on DDS entity.
```

And below is the size of the data I am currently sending.

julianbermudez, 2019-05-16:
Hi @OpusK, 

I was testing your issue and it seems that you may have not set properly the `MTU` in the Agent.
Please, try to set the CMake flag `CONFIG_UDP_TRANSPORT_MTU` to a large enough value in the Agent, and use the `<historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>` property in both DataWriter and DataReader XML.
Please let me know if it solves your issue.

OpusK, 2019-05-17:
Hi, @julianbermudez 

I am currently using the following options.

```
cmake -DTHIRDPARTY=ON -DCONFIG_UDP_TRANSPORT_MTU=4096 -DCONFIG_SERIAL_TRANSPORT_MTU=4096 ..
```

Currently, I have set the XML of DataWriter as below and use echo function of CLI of ROS2 as DataReader role (Is it possible to modify XML of DataReader in this case?).

```
    <data_writer profile_name="scan_half">
        <topic>
            <kind>NO_KEY</kind>
            <name>rt/scan_half</name>
            <dataType>sensor_msgs::msg::dds_::LaserScan_</dataType>
        </topic>
        <publisher>
            <historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
        </publisher>
    </data_writer>
```

OpusK, 2019-05-17:
@julianbermudez 

I confirmed that it works normally on a PC using b3c43bd20b16bd3570a128f05a9bf1a164883435-commit version.

closed 2019-05-17 by OpusK