eProsima/Micro-XRCE-DDS-Client #94: Compile Error

OpusK, 2019-06-11:
Hi,

A compilation error occurs in log.c.
If I exclude log.c from the compilation there is no problem, but anyway, I tell you that client has this problem.

```
D:\source\arduino_sketchbook\libraries\ros2arduino\src\ros2\xrcedds\micro_xrce_dds\lib\src\c\core\log\log.c: In function 'print_create_client_submessage':

D:\source\arduino_sketchbook\libraries\ros2arduino\src\ros2\xrcedds\micro_xrce_dds\lib\src\c\core\log\log.c:380:39: error: 'CREATE_CLIENT_Payload {aka const struct CREATE_CLIENT_Payload}' has no member named 'base'

             request_to_string(&payload->base),

                                       ^

D:\source\arduino_sketchbook\libraries\ros2arduino\src\ros2\xrcedds\micro_xrce_dds\lib\src\c\core\log\log.c: In function 'print_status_agent_submessage':

D:\source\arduino_sketchbook\libraries\ros2arduino\src\ros2\xrcedds\micro_xrce_dds\lib\src\c\core\log\log.c:486:37: error: 'STATUS_AGENT_Payload {aka const struct STATUS_AGENT_Payload}' has no member named 'base'

             reply_to_string(&payload->base),

                                     ^

D:\source\arduino_sketchbook\libraries\ros2arduino\src\ros2\xrcedds\micro_xrce_dds\lib\src\c\core\log\log.c: In function 'print_read_data_submessage':

D:\source\arduino_sketchbook\libraries\ros2arduino\src\ros2\xrcedds\micro_xrce_dds\lib\src\c\core\log\log.c:539:40: error: 'ReadSpecification {aka const struct ReadSpecification}' has no member named 'input_stream_id'

             payload->read_specification.input_stream_id,

                                        ^
```

Actually, the member does not exist as shown below.

[struct ReadSpecification](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/master/src/c/core/serialization/xrce_protocol_internal.h#L671)

julianbermudez, 2019-06-11:
Hi @OpusK,

This issue should be fixed in pull request #95. Could you check it? Thanks!

OpusK, 2019-06-12:
@julianbermudez 

Thank you! It is resolved.
I will ask you one more thing.

For log.c to line [657](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/9fc878bc961f7664f87af6c9206d655ece77f6cd/src/c/core/log/log.c#L657), a warning will be issued, and depending on the compiler option, the compilation will fail.
We will request this warning processing. (like below or any way)

```
printf(" %st: %lims%s", BLUE, (long int)ms, RESTORE_COLOR);
```

julianbermudez, 2019-06-12:
Hi @OpusK,

I tried to fix this warning using the `PRI**` approach, I hope it helps you.

OpusK, 2019-06-12:
Hi, @julianbermudez 

Resolved!
Thank you :)

closed 2019-06-12 by OpusK