eProsima/Micro-XRCE-DDS-Client #100: Defaults value for CONFIG_MIN_SESSION_CONNECTION_INTERVAL is too low for reliable TCP stream

sp193, 2019-08-31:
Hi,

I believe the default value for ``CONFIG_MIN_SESSION_CONNECTION_INTERVAL`` is too low.
By default, ``CONFIG_MIN_SESSION_CONNECTION_INTERVAL`` is 1 (ms) and ``CONFIG_MAX_SESSION_CONNECTION_ATTEMPTS`` is 12. This means that a XRCE client may try up to 12 times, with the maximum waiting period beginning from 1ms.

The sequence would be like this, assuming a successful transmission takes place and no response is received within the window:
``1, 2, 4, 8, 16, 32, 64, 128, 256, 512, 1024, 2048``

If a message is received but is not a correct response, then it gets worse because the client will continue to wait for only 1ms.

However, depending on the speed of the client, the agent and the order of execution,  the agent might get multiple ``CREATE SESSION ``messages from the client. This could be dangerous, if the agent successfully resets a reliable stream repeatedly, after at least one (reply) message has been sent. Resetting a reliable output stream after the session has been created, could result in a communications breakdown with no recovery, due to the sequence numbers going wrong.

The line in question is [here](https://github.com/sp193/Micro-XRCE-DDS-Agent/blob/master/src/cpp/Root.cpp#L108):
```
                {
                    client->session().reset();
                }
```

If you add print statements/set a breakpoint before this line, it is possible to occasionally observe a session getting reset multiple times, for a client.

Across the Internet, it is possible for the RTT to be more than even 5ms. If TCP is used and Nagle is enabled, then the message could be delayed for up to 250ms, particularly if it is a small message.

In my opinion, I think increasing ``CONFIG_MIN_SESSION_CONNECTION_INTERVAL`` to a larger value like 500 could help reduce the odds of the agent receiving duplicate ``CREATE SESSION`` requests.
In exchange,`` CONFIG_MIN_SESSION_CONNECTION_INTERVAL`` could be reduced to 5, to prevent the timeout period from increasing to huge numbers.

julianbermudez, 2019-09-02:
Hi @sp193,

We appreciate your piece of advice. We will take it into account this as the default configuration for new releases. Thanks.

BorjaOuterelo, 2019-12-03:
Hi @sp193 this has been addressed in https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/110.

closed 2019-12-03 by BorjaOuterelo