eProsima/Micro-XRCE-DDS-Client #102: examples/Discovery/main.c:49:33: error: assignment to expression with array type  agent_list[size].ip = argv[i];

yottayuan, 2019-10-08:
follow:https://micro-xrce-dds.readthedocs.io/en/latest/installation.html#installing-the-client-stand-alone
------------
lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.3 LTS
Release:	18.04
Codename:	bionic
---------------
apt-cache show ros-dashing-ros-base 
Package: ros-dashing-ros-base
Version: 0.7.2-1bionic.20190923.212341
Architecture: amd64
-----
edit: client.config file: Add:
PROFILE_CREATE_ENTITIES_XML=TRUE
PROFILE_READ_ACCESS=TRUE
PROFILE_WRITE_ACCESS=TRUE
-----
cmake -DUCLIENT_BUILD_EXAMPLES=ON ..
make
it appears the error:
/home/yottayuan/Micro-XRCE-DDS-Client/examples/Discovery/main.c: In function ‘main’:
/home/yottayuan/Micro-XRCE-DDS-Client/examples/Discovery/main.c:49:33: error: assignment to expression with array type
             agent_list[size].ip = argv[i];
                                 ^
examples/Discovery/CMakeFiles/Discovery.dir/build.make:62: recipe for target 'examples/Discovery/CMakeFiles/Discovery.dir/main.c.o' failed
make[2]: *** [examples/Discovery/CMakeFiles/Discovery.dir/main.c.o] Error 1
CMakeFiles/Makefile2:420: recipe for target 'examples/Discovery/CMakeFiles/Discovery.dir/all' failed
make[1]: *** [examples/Discovery/CMakeFiles/Discovery.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

Thanks.

julianbermudez, 2019-10-09:
Hi @yottayuan,

This issue has been fixed.

yottayuan, 2019-10-09:
@julianbermudez Thanks for the fix,it works now!

closed 2019-10-09 by yottayuan