eProsima/Micro-XRCE-DDS-Client #150: Any plans to support the Arduino library?

OpusK, 2020-05-14:
Hi.

Are there plans to support the Arduino library for Micro-XRCE-DDS-Client?
Currently, it is cloned/modified by ros2arduino, but I thought it would be better if there was an officially managed library. (Simply not ros2 only)

JaimeMartin, 2020-05-14:
Hi @OpusK , @julibert 

Did we evaluate the possibility to use Micro-ROS for this? I think it would be nice to meet and discuss potential options to support Arduino better.

OpusK, 2020-05-14:
Hi, @JaimeMartin.

I have no experienced Micro-ROS.
But, what I asked is the opinion about support as an XRCE-DDS Client not related to ROS.
It was about support as middleware itself.
Of course, this was not an Arduino build support using Cmake, but a question about official library support registered with Arduino's library manager.

closed 2020-07-09 by OpusK