eProsima/Micro-XRCE-DDS-Client #158: can't see topic with the ros2 topic list command

szc188, 2020-06-03:
Hi.
I'm using the PublishHelloWorld sample code.
I have a problem getting topic list.

**Environment**
Ubuntu 18.04.4 LTS
Micro-XRCE-DDS-Agent v1.4.0
Micro-XRCE-DDS-Client v1.2.3
ROS Dashing Patch6

**Problem**
```
➜  ~ ros2 node list
/default_xrce_participant

➜  ~ ros2 topic list
/parameter_events
/rosout
```

Do you have any idea on how to solve this problem?

Thank you.

szc188, 2020-06-03:
**PublishHelloWorld**
```
➜  PublishHelloWorld git:(master) ./PublishHelloWorldClient localhost 2018
Send topic: Hello DDS world!, id: 1
Send topic: Hello DDS world!, id: 2
Send topic: Hello DDS world!, id: 3
Send topic: Hello DDS world!, id: 4
Send topic: Hello DDS world!, id: 5
Send topic: Hello DDS world!, id: 6
Send topic: Hello DDS world!, id: 7
Send topic: Hello DDS world!, id: 8
Send topic: Hello DDS world!, id: 9
Send topic: Hello DDS world!, id: 10
^C
```

**SubscribeHelloWorld**
```
➜  SubscribeHelloWorld git:(master) ./SubscribeHelloWorldClient localhost 2018
Received topic: Hello DDS world!, id: 2
Received topic: Hello DDS world!, id: 3
Received topic: Hello DDS world!, id: 4
Received topic: Hello DDS world!, id: 5
Received topic: Hello DDS world!, id: 6
Received topic: Hello DDS world!, id: 7
Received topic: Hello DDS world!, id: 8
Received topic: Hello DDS world!, id: 9
Received topic: Hello DDS world!, id: 10
```

**Agent**
```
➜  build git:(d444f99) ls
agent.refs  CMakeCache.txt  cmake_install.cmake  libmicroxrcedds_agent.so      libmicroxrcedds_agent.so.1.4.0  MicroXRCEAgent  uagent-prefix
cmake       CMakeFiles      include              libmicroxrcedds_agent.so.1.4  Makefile                        sanitizers
➜  build git:(d444f99) ./MicroXRCEAgent udp4 -p 2018 -v 6      
Press CTRL+C to exit
[1591152618.106063] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 2018
[1591152618.106205] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1591152682.800038] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F CC CC DD DD 81 00 FC 01
[1591152682.800241] info     | Root.cpp           | create_client            | create                 | client_key: 0xCCCCDDDD, session_id: 0x81
[1591152682.800300] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x3435978205, address: 127.0.0.1:60117
[1591152682.800373] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xCCCCDDDD, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1591152682.800475] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xCCCCDDDD, len: 388, data: 
0000: 81 80 00 00 01 05 68 00 00 0A 00 11 01 02 00 00 59 00 00 00 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 64 65 66 61 75 6C 74 5F 78 72 63 65 5F 70
0040: 61 72 74 69 63 69 70 61 6E 74 3C 2F 6E 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63
0060: 69 70 61 6E 74 3E 3C 2F 64 64 73 3E 00 00 00 00 01 05 64 00 00 0B 00 12 02 02 00 00 56 00 00 00
0080: 3C 64 64 73 3E 3C 74 6F 70 69 63 3E 3C 6E 61 6D 65 3E 48 65 6C 6C 6F 57 6F 72 6C 64 54 6F 70 69
00A0: 63 3C 2F 6E 61 6D 65 3E 3C 64 61 74 61 54 79 70 65 3E 48 65 6C 6C 6F 57 6F 72 6C 64 3C 2F 64 61
00C0: 74 61 54 79 70 65 3E 3C 2F 74 6F 70 69 63 3E 3C 2F 64 64 73 3E 00 00 11 01 05 0F 00 00 0C 00 14
00E0: 04 02 00 00 01 00 00 00 00 00 11 00 01 05 92 00 00 0D 00 16 06 02 00 00 84 00 00 00 3C 64 64 73
0100: 3E 3C 64 61 74 61 5F 72 65 61 64 65 72 3E 3C 74 6F 70 69 63 3E 3C 6B 69 6E 64 3E 4E 4F 5F 4B 45
0120: 59 3C 2F 6B 69 6E 64 3E 3C 6E 61 6D 65 3E 48 65 6C 6C 6F 57 6F 72 6C 64 54 6F 70 69 63 3C 2F 6E
0140: 61 6D 65 3E 3C 64 61 74 61 54 79 70 65 3E 48 65 6C 6C 6F 57 6F 72 6C 64 3C 2F 64 61 74 61 54 79
0160: 70 65 3E 3C 2F 74 6F 70 69 63 3E 3C 2F 64 61 74 61 5F 72 65 61 64 65 72 3E 3C 2F 64 64 73 3E 00
0180: 00 14 00 00
[1591152682.801599] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xCCCCDDDD, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1591152682.802576] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0xCCCCDDDD, participant_id: 0x001(1)
[1591152682.802687] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xCCCCDDDD, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 11 00 00
[1591152682.802760] debug    | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0xCCCCDDDD, topic_id: 0x001(2), participant_id: 0x001(1)
[1591152682.802784] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xCCCCDDDD, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 12 00 00
[1591152682.802812] debug    | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0xCCCCDDDD, subscriber_id: 0x001(4), participant_id: 0x001(1)
[1591152682.802843] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xCCCCDDDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[... 182 lines cut ...]
0000: 09 00 00 00 11 00 00 00 48 65 6C 6C 6F 20 44 44 53 20 77 6F 72 6C 64 21 00
[1591152718.072777] debug    | DataReader.cpp     | read_fn                  | [==>> DDS <<==]        | client_key: 0x00000001, len: 25, data: 
0000: 09 00 00 00 11 00 00 00 48 65 6C 6C 6F 20 44 44 53 20 77 6F 72 6C 64 21 00
[1591152718.072826] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xAAAABBBB, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 0A 00 00 00 80
[... cut ...]
0000: 81 00 00 00 0A 01 05 00 0B 00 00 00 80
[1591152719.073631] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xCCCCDDDD, len: 37, data: 
0000: 81 80 0C 00 09 01 1D 00 00 0E 00 16 0A 00 00 00 11 00 00 00 48 65 6C 6C 6F 20 44 44 53 20 77 6F
0020: 72 6C 64 21 00
[1591152719.073792] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xCCCCDDDD, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 0D 00 00 00 80
```

pablogs9, 2020-06-03:
Hello @szc188, name mangling in Micro-XRCE-DDS is not compatible with ROS2 by default. Please read [this](https://design.ros2.org/articles/topic_and_service_names.html).

You should be careful of topic and types names, for example a publisher of 32 bits integer in Micro XRCE-DDS that is compatible with ROS2 looks like (in the XML approach):

```
    <data_writer profile_name="weather_station/temperature__dw">
        <historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
        <qos>
          <reliability>
            <kind>RELIABLE</kind>
          </reliability>
        </qos>
        <topic>
            <kind>NO_KEY</kind>
            <name>rt/weather_station/temperature</name>
            <dataType>std_msgs::msg::dds_::Float32_</dataType>
        </topic>
    </data_writer>
```

If you want to be compatible by default, you can check (micro-ROS)[https://github.com/micro-ROS/micro-ros-build/tree/dashing/micro_ros_setup], a ROS 2 port for embedded systems that uses Micro XRCE-DDS as the default middleware layer.

szc188, 2020-06-03:
Hi @pablogs9 

Thank you.
It is good working!

closed 2020-06-03 by szc188