eProsima/Micro-XRCE-DDS-Client #159: Allow external TCP/UDP support for `udp_transport_datagram_internal`

driftregion, 2020-06-08:
Hello,
I'm building for a FreeRTOS-based platform using FreeRTOS-Plus-TCP.  Its socket API lacks the functions and structs defined in linux's `<sys/socket.h>` .

My toolchain file contains:
```
set(CMAKE_SYSTEM_NAME Generic)
```
I'm able to make use of the sensible defaults for UDP and TCP defined in `udp`/`tcp_transport_internal.h` for the transport functionality.  I'm compiling Micro-XRCE-DDS-Client as a static library.  The implementations of the stubbed functions (using FreeRTOS-Plus-TCP) are inside my application code.  All is well at link time.

I can't do the same thing for `udp_transport_datagram_internal.h`, because it defines `uxrUDPTransportDatagram` such that there's no means for extension

I propose something like this:
```
typedef struct uxrUDPTransportDatagram
{
    uint8_t buffer[UXR_UDP_TRANSPORT_MTU_DATAGRAM];
#if defined(UCLIENT_PLATFORM_POSIX)
    struct pollfd poll_fd;
#elif defined(UCLIENT_PLATFORM_WINDOWS)
    WSAPOLLFD poll_fd;
#else
    void *poll_fd;
#endif
```

julibert, 2020-06-08:
Hi @driftregion,

Yes, it could be a short-term solution, but I think a more appropriate way is to handle FreeRTOS platform similar to [nuttx](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/fc334262e50e89d4fd874a9570a6482f9f202ba9/CMakeLists.txt#L118) or [LwIP](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/fc334262e50e89d4fd874a9570a6482f9f202ba9/CMakeLists.txt#L120). For this purpose the following changes are required:

- [ ] Modify CMakeLists.txt to support `freertos` `PLATFORM_NAME`.
- [ ] Move `uxrUDPTransportDatagram`  type declaration to `udp_transport_datagram_posix.c` and `udp_transport_datagram_windows.c`.
- [ ] Create a `udp_transport_datagram_freertos.c` with its own `uxrUDPTransportDatagram` type (it seems that FreeRTPS-Plus-TCP` has a [pollfd](https://github.com/FreeRTOS/FreeRTOS/blob/cb7edd2323a77f3dbea144c1f48f95582becc99e/FreeRTOS-Plus/Source/FreeRTOS-Plus-TCP/FreeRTOS_Sockets.c#L3974) structure.

You can purpose the short-term solution by a PR and we will create an issue in order to put in the roadmap the long-term solution.

driftregion, 2020-06-08:
Got it, thanks @julibert 
I wrongly assumed that supporting FreeRTOS-Plus-TCP was out of scope.  I've got a workaround for this now.  Let's work towards the long-term solution.

driftregion, 2020-06-12:
I'm attempting the long-term solution.
Question: does LwIP guarantee a `clock_gettime`?  I'm curious where that comes from when compiling for LwIP targets.

FranFin, 2020-07-20:
@driftregion Hi! Thank you for your interest in Micro XRCE-DDS. Notice that we’ll discuss the recent contributions and developments in the next [Embedded WG meeting](https://discourse.ros.org/t/ros-2-embedded-wg-meetings/15460). You’re invited to bring your Micro XRCE-DDS related issue to this platform. We’ll be very happy to have you join us and discuss your questions with the community.