eProsima/Micro-XRCE-DDS-Client #160: Compile error

Yoke365, 2020-06-11:
# ~/env/Micro-XRCE-DDS-Client/build$ make 
Scanning dependencies of target microcdr
[  6%] Creating directories for 'microcdr'
[ 12%] Performing download step (git clone) for 'microcdr'
Cloning into 'microcdr'...
Note: checking out 'v1.2.0'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 6c9a88f... Release v1.2.0 (#52)
[ 18%] No patch step for 'microcdr'
[ 25%] Performing update step for 'microcdr'
[ 31%] Performing configure step for 'microcdr'
loading initial cache file /home/pv/env/Micro-XRCE-DDS-Client/build/microcdr/tmp/microcdr-cache-.cmake
-- Setting build type to 'Release' as none was specified.
-- The C compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- The CXX compiler identification is GNU 5.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pv/env/Micro-XRCE-DDS-Client/build/microcdr/src/microcdr-build
[ 37%] Performing build step for 'microcdr'
Scanning dependencies of target ucdr
[ 12%] Creating directories for 'ucdr'
[ 25%] No download step for 'ucdr'
[ 37%] No patch step for 'ucdr'
[ 50%] No update step for 'ucdr'
[ 62%] Performing configure step for 'ucdr'
loading initial cache file /home/pv/env/Micro-XRCE-DDS-Client/build/microcdr/src/microcdr-build/ucdr-prefix/tmp/ucdr-cache-Release.cmake
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pv/env/Micro-XRCE-DDS-Client/build/microcdr/src/microcdr-build
[ 75%] Performing build step for 'ucdr'
Scanning dependencies of target microcdr
[ 16%] Building C object CMakeFiles/microcdr.dir/src/c/common.c.o
[ 33%] Building C object CMakeFiles/microcdr.dir/src/c/types/basic.c.o
[ 50%] Building C object CMakeFiles/microcdr.dir/src/c/types/string.c.o
[ 66%] Building C object CMakeFiles/microcdr.dir/src/c/types/array.c.o
[ 83%] Building C object CMakeFiles/microcdr.dir/src/c/types/sequence.c.o
[100%] Linking C static library libmicrocdr.a
[100%] Built target microcdr
No install step for 'ucdr'
Completed 'ucdr'
Built target ucdr
[ 43%] Performing install step for 'microcdr'
[100%] Built target microcdr
Install the project...
-- Install configuration: "Release"
-- Installing: /home/pv/env/Micro-XRCE-DDS-Client/build/temp_install/microcdr-1.2.0/lib/libmicrocdr.a
-- Installing: /home/pv/env/Micro-XRCE-DDS-Client/build/temp_install/microcdr-1.2.0/include/ucdr
-- Installing: /home/pv/env/Micro-XRCE-DDS-Client/build/temp_install/microcdr-1.2.0/include/ucdr/microcdr.h
-- Installing: /home/pv/env/Micro-XRCE-DDS-Client/build/temp_install/microcdr-1.2.0/include/ucdr/visibility.h
-- Installing: /home/pv/env/Micro-XRCE-DDS-Client/build/temp_install/microcdr-1.2.0/include/ucdr/config.h
-- Installing: /home/pv/env/Micro-XRCE-DDS-Client/build/temp_install/microcdr-1.2.0/share/microcdr/cmake/microcdrTargets.cmake
-- Installing: /home/pv/env/Micro-XRCE-DDS-Client/build/temp_install/microcdr-1.2.0/share/microcdr/cmake/microcdrTargets-release.cmake
-- Installing: /home/pv/env/Micro-XRCE-DDS-Client/build/temp_install/microcdr-1.2.0/share/microcdr/cmake/microcdrConfig.cmake
-- Installing: /home/pv/env/Micro-XRCE-DDS-Client/build/temp_install/microcdr-1.2.0/share/microcdr/cmake/microcdrConfigVersion.cmake
[ 50%] Completed 'microcdr'
[ 50%] Built target microcdr
Scanning dependencies of target uclient
[ 56%] Creating directories for 'uclient'
[ 62%] No download step for 'uclient'
[ 68%] No patch step for 'uclient'
[ 75%] No update step for 'uclient'
[ 81%] Performing configure step for 'uclient'
loading initial cache file /home/pv/env/Micro-XRCE-DDS-Client/build/uclient-prefix/tmp/uclient-cache-.cmake
-- Setting build type to 'Release' as none was specified.
CMake Error at CMakeLists.txt:106 (find_package):
  By not providing "Findmicrocdr.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "microcdr",
  but CMake did not find one.

  Could not find a package configuration file provided by "microcdr"
  (requested version 1.2.0) with any of the following names:

    microcdrConfig.cmake
    microcdr-config.cmake

  Add the installation prefix of "microcdr" to CMAKE_PREFIX_PATH or set
  "microcdr_DIR" to a directory containing one of the above files.  If
  "microcdr" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/pv/env/Micro-XRCE-DDS-Client/build/CMakeFiles/CMakeOutput.log".
CMakeFiles/uclient.dir/build.make:107: recipe for target 'uclient-prefix/src/uclient-stamp/uclient-configure' failed
make[2]: *** [uclient-prefix/src/uclient-stamp/uclient-configure] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/uclient.dir/all' failed
make[1]: *** [CMakeFiles/uclient.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

pablogs9, 2020-06-11:
Please could you write the steps you have followed in order to reproduce this issue?

Yoke365, 2020-06-30:
> Please could you write the steps you have followed in order to reproduce this issue?

Follow this https://micro-xrce-dds.docs.eprosima.com/en/latest/installation.html.

```
 git clone https://github.com/eProsima/Micro-XRCE-DDS.git
 cd Micro-XRCE-DDS
 mkdir build && cd build
 cmake ..
 make
 sudo make install
```

pablogs9, 2020-06-30:
Please install client and agent in separate folders:

```
git clone -b v1.4.2 https://github.com/eProsima/Micro-XRCE-DDS-Agent
cd Micro-XRCE-DDS-Agent 
mkdir build && cd build
cmake ..
make
```

and

```
git clone -b v1.2.5 https://github.com/eProsima/Micro-XRCE-DDS-Client
cd Micro-XRCE-DDS-Client
mkdir build && cd build
cmake .. -DUCLIENT_BUILD_EXAMPLES=ON
make
```

closed 2020-10-19 by pablogs9