eProsima/Micro-XRCE-DDS-Client #168: compiling for new transport

vibnwis, 2020-10-15:
Hi Pablo,
I think I am stuck with the compilation. I had run the command, ros2 run micro_ros_setup create_firmware_ws.sh freertos olimex-stm32-e407 of which created the firmware folder. For my case, 

I make changes on some files in the folder firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/PublishHellowWorld/main.c
, and added CAN_circular_buffer.c, CAN_circular_buffer.h.

Also, a file I changed in the folder Micro-XRCE-DDS-Client/src/c/profile/transport/ip/udp/udp_transport.c 

I have not changed those variables in the CMakeList.txt relating to the static memory allocation yet. I will leave at a later stage when I have passed the compilation and debugging stages.

I followed almost to the completion of the step in https://micro-ros.github.io/docs/tutorials/core/create_new_type/, but I think I have got into a wrong pathway, as I am not creating a new message. I am creating a transport, but still using existing uxr_xxx_read()/write() functions, and using the main.c in the folder firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/PublishHellowWorld.

I have also read the other issue with regards to compile errors. I was wondering that is applicable to me too, i.e., I should follow the instructions there, should I. I will give it a go while waiting for your answers.

Appreciate your advice, please.

vibnwis, 2020-10-15:
Okie, I followed your instructions. as shown below, on the other issue. It works. Now I am resolving the error messages.

git clone -b v1.2.5 https://github.com/eProsima/Micro-XRCE-DDS-Client
cd Micro-XRCE-DDS-Client
mkdir build && cd build
cmake .. -DUCLIENT_BUILD_EXAMPLES=ON
make

pablogs9, 2020-10-15:
I just have answered you [here](https://github.com/micro-ROS/micro_ros_setup/issues/206#issuecomment-708910458).

Maybe you can create a fork of this repo with your changes and we can discuss here how to implement your transport.

vibnwis, 2020-10-15:
Sounds good for the suggestion.

Many thanks for the solution. I have got it compiled with the changes I
made. However, it does not recognise the new files, I presume the Makefile
needs to be edited to incorporate that. BTW, since the
eProsima/Micro-XRCE-DDS-Client <
https://github.com/eProsima/Micro-XRCE-DDS-Client> is also exist in
firmware/mcu_sw/eProsima/Micro-XRCE-DDS-Client <
https://github.com/eProsima/Micro-XRCE-DDS-Client>, I would not need to git
that separately or I need to for some reasons? Of course, the final
question, how do I run or specify what to run those examples ?

pablogs9, 2020-10-15:
Of course, both the serialization (CDR) and the middleware (XRCE client) are present in the micro-ROS workspace because they are used by micro-ROS. But I suggest you building them completely apart of micro-ROS in order to ease your transport development. As I said previously, once the CAN transport is ready with the XRCE middleware it can be easily integrated in micro-ROS.

Regarding the modifications: remember that we are using a CMake project not a Make one. So in order to include files you should add them to the CMakeLists.txt: [check here](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/66df0a6df20063d246dd638ac3d33eb2e652fab2/CMakeLists.txt#L136).

Do you have any repo with the CAN transport that I can see?

vibnwis, 2020-10-15:
The source is very rough at the moment. If things go well, it should be ready by tomorrow.

vibnwis, 2020-10-15:
"But I suggest you building them completely apart of micro-ROS in order to ease your transport development." I presume you mean, git down the XRCE client and compile it in a separate folder. While typing this, XRCE Client is being compiled. I can change those intended files accordingly. However, would you please share how to test it, and specifying what to run? also I presume it is running on a target board, i.e., Olimex STM32-E407 for my case. Sorry, I seriously can't "very far" of the entire mircoROS yet. Hence, be patient with my ignorance.

pablogs9, 2020-10-15:
Is possible for you to run your CAN library in your host computer? The same library that you are going to use in the embedded platform?

If yes, we can test it directly with the XRCE-DDS library.

If not, we need to test it directly in the board. You have [this repo](https://github.com/eProsima/Micro-XRCE-DDS-Apps/tree/feature/refactor_nuttx/FreeRTOS) in order to build just the XRCE-DDS Client in the Olimex board with FreeRTOS. I think that this can be useful to test the CAN code. It follows the same structure of a STMCubeMX proyect.

vibnwis, 2020-10-15:
My host computer does not have the CAN controller as well as the
transceiver. Hence, I can not run it. I think the suggested idea makes more
sense.

However, I presume I still need to build the CAN transport using the
udp_transport_external.c.template in eProsima/Micro-XRCE-DDS-Client ? Also
I presume, somehow, in XRCE-DDS Client can call the CAN-transport
accordingly.

pablogs9, 2020-10-15:
Yes, of course using  [this repo](https://github.com/eProsima/Micro-XRCE-DDS-Apps/tree/feature/refactor_nuttx/FreeRTOS) you should be able to integrate your transport in the XRCE-DDS Client library.

vibnwis, 2020-10-15:
I am wanting to test the comm. of two STM32-E407 boards through CAN bus. it derived from the UDP PingPong demo, that CAN-bus being a transport protocol should work.  Reading the MicroROS architecture, I was wondering where does the XRCE-DDS agent reside in this case? Should it be present as part of the system?

pablogs9, 2020-10-15:
The micro-ROS Agent should be installed in a Linux/Windows computer or it a Raspberry Pi. Both Olimexs should be connected to this computer's Agent and they will be able to talk to each other and to the whole ROS 2 world.

vibnwis, 2020-10-15:
Fortunately I asked that. Unfortunately, I do not or have not developed the CAN driver for Raspberry Pi nor Linux/Windows. I guess I would need to source/buy a dongle (USB) which has a CAN driver ;-(

A long shot, any possibilities or adjustment that testing the comm without micro-ROS Agent?

vibnwis, 2020-10-15:
Since the uROS Agent resides in PC, and uROS Client resides in STM32
boards, I was wondering where the respective compilers are being set? I
didn't set them by far.

pablogs9, 2020-10-15:
The micro-ROS Client is being crosscompiled for the hardware platform (usually the `micro_ros_setup` package takes cares of this) and the micro-ROS Agent is being compiled for the host computer with the usual Linux/Windows compiler (in the case of the agent for Raspberry Pi you also have the option to crosscompile the agent using the `micro_ros_setup` ).

vibnwis, 2020-10-15:
The reason I asked that was because I am now compile it separately from microROS, both the agent and the client. With this setting, I presume the agent and the client do not have any knowledge of local setting, but only the default setting attached to the repo. Is it right to view it like this?

vibnwis, 2020-10-15:
Local, I meant the setting derived from microRos. microROS is not in the picture since I compile them separately.

pablogs9, 2020-10-16:
Sorry, I didn't understood your question. Could you be more specific?

vibnwis, 2020-10-16:
I am compiling the XRCE-client in particular separately, without using
MicroRos setup, neither I use any command to set the target board.  Hence,
it raises a question of what board it is for? Where do you set it? The same
applies to XRCE-agent for Raspberry Pi on which I will need to set it up.
However, I presume for XRCE-agent since it is using the native compiler,
the default settings apply.

Hope this help.

vibnwis, 2020-10-16:
Watched the DDS-RTPS, XRCE-DDS and microROS presentation video. It is very
impressive, and my interest in learning is now growing.

vibnwis, 2020-10-17:
Hi Pablo,
I am kind of stuck with the something, your advice would indeed enlighten me.
I am now compiling the micro-XRCE-DDS-agent on Raspberry Pi 4. It should go well, hopefully. Of course, of my case, it still needs the CAN driver, and I am getting it installed soon the compilation completed. Would you mind sharing with the CAN driver availability, in micro-XRCE-Agent, is there a template for implementing local transports? please elaborate if it is different from the Client. please?

vibnwis, 2020-10-19:
My bad, I missed out the lines "... You have this repo in order to build just the XRCE-DDS Client in the Olimex board with FreeRTOS..". Apologies.

closed 2020-10-19 by vibnwis