eProsima/Micro-XRCE-DDS-Client #169: CAN-bus transport coding for micro-XRCE-Client and Agent

vibnwis, 2020-10-18:
Hi Pablo,
It would be very appreciated you could share your opinions on where-about the code for CAN-bus transport should be placed.

1. I will use the CAN-RX interrupt, and place them incoming bytes into a circular buffer for the uxr_xx_read() and uxr_xx_write() in udp_transport.c. As such, STM32CubeMX will be used to create the skeleton of the CAN-RX interrupt, and its, the interrupt body will fill up the circular buffer from which the uxr_xx_read() will be reading. For the same reason, the CAN-TX function will also reside in the stm32-e407 main(), and uxr_xxx_write() will call it, the CAN_Tx() when transmitting data. I think microROS which has the controller's low-level files, is needed. Does it make sense?

2. I have read about CANSocket, which is a replica of Network Socket. With that I think I can change the body of UDPv4AgentLinux.cpp, i.e., bool UDPv4Agent::init(),  bool UDPv4Agent::fini()
bool UDPv4Agent::send_message(), 
bool UDPv4Agent::recv_message(). I would like to seek your opinion on that, please?

3. BTW, can you please look into the read messages when I compiling in Raspberry Pi4? FYI, I did not specify it was Raspberry. I was wondering if that is the cause of the errors.

I look forward to hearing from you.

Thank you
V!

vibnwis, 2020-10-19:
I guess, the case #1 is no longer valid, as it has been answered in the other issues. However, any comments of case #2 and #3 would be appreciated.
Thank you

pablogs9, 2020-10-19:
Hello,

1. Share code if possible
2. I have not implemented any custom transport in Micro XRCE-DDS Agent, so I'm not sure about the feasibility of rewriting transports like this. Maybe @jamoralp can help us here.
3. Answered [here](https://github.com/eProsima/Micro-XRCE-DDS-Agent/issues/192)

vibnwis, 2020-10-19:
Sure! I will, very soon.

jamoralp, 2020-12-21:
Closing this since inactive for two months. Please continue the discussion in the issue of the Agent repo, and slack channel

closed 2020-12-21 by jamoralp