eProsima/Micro-XRCE-DDS-Client #173: Does TimeSync example work correctly?

liyue75, 2020-10-29:
When I ran the TimeSync example, I found the result like this:
synchronized with time offset 11187us
synchronized with time offset 411497us
sychronized with time offset 22794us
sychronized with time offset 471705us
...
I ran an Micro-XRCE-DDS-agent on an computer and a client on another computer, both of them were on the same LAN.
(I made some change in CMakelists and main.c of TimeSync to let it build and compile)
Why does time offset keep changing? Shouldn't the two computers have fixed time offset?

pablogs9, 2020-10-29:
Hello @liyue75, thanks for using out XRCE-DDS library.

Regarding your results, we have been testing the time sync implementation. You can find the main client part [here](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/66df0a6df20063d246dd638ac3d33eb2e652fab2/src/c/core/session/session.c#L744) and the main server part [here](https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/559952b5fd47f00db4336ae8eb078af8a1e5d1f5/src/cpp/processor/Processor.cpp#L674). As you can see by default we are implementing a basic NTP, you can find more details [here](https://en.wikipedia.org/wiki/Network_Time_Protocol#Clock_synchronization_algorithm).

As you can see this is not a super high precision time synchronization algorithm and the order of hundreds of milliseconds of error that you are obtaining seems to be expected.

Using this same approach, the XRCE-DDS Client library also offers a callback that can be set with `uxr_set_time_callback(&session, on_time, NULL)`. Using this callback you can code your own averaging mechanism like the one found [here](https://micro-xrce-dds.docs.eprosima.com/en/latest/time_sync.html) (see the last code snippet). Using this averaging procedure I obtain a more stable time offset between the client and the agent.

Let me know your thoughts.

liyue75, 2020-10-30:
Hi @pablogs9, thank you for your reply.
I use the sync example because I want to synchronise the time of client which is running on a stm32 board with agent on computer.
Since I pretend to test the latecy in udp transfer between two clients  running on stm32 boards, so I want to get the time offset between these boards and computer first and then synchronise their time.
But it seems difficult to implement it if I want to get accurate time offset in software, now I plan to synchronise stm32boards  in hardware.

pablogs9, 2020-10-30:
Ok, hope this info has helped. Are you using bare XRCE-DDS Client or micro-ROS? 

I'm closing this, do not hesitate to reopen if you have any other question.

Thanks!

closed 2020-10-30 by pablogs9