eProsima/Micro-XRCE-DDS-Client #181: Best effort client not showing topics even though Agent is connected

eranroll, 2021-01-24:
Hi,

I am trying to get a client working on an embedded device without an OS, which has an agent running over linux.
When I use "reliable" QoS everything works as expected, however, when I try to use "best effort" I can't seem to find any topics. Even though the Agent shows logs as if the connection is alive.

Any modification required on the Agent side? Any elaborate examples for such a use case?

Any help is much appreciated,
Regards

pablogs9, 2021-01-25:
How are you trying to see these "best-effort" topics? Remember that DDS QoS must match between the publisher and the subscriber. That is that a reliable subscriber won't receive topics from a best-effort publisher.

Please provide more details.

eranroll, 2021-01-25:
I am actually using the topics as ROS topics. I have added the required prefixes, and when using Reliable, it all works.
I am doing the following:
1. Launch an XRCE Agent over serial
2. Boot the XRCE Client
3. Once it has booted - ros2 topic list - should show the topics.

It shows it on reliable, but not when I try to use best effort

pablogs9, 2021-01-25:
Could you share the XML profile that you are using for these best effort publishers?

eranroll, 2021-01-25:
I believe these are similar to the default XMLs:

#define UXR_PARTICIPANT_XML "<dds><participant><rtps><name>%s</name></rtps></participant></dds>"
#define UXR_WRITER_XML "<dds><data_writer><topic><kind>NO_KEY</kind><name>%s</name><dataType>%s</dataType></topic></data_writer></dds>"
#define UXR_READER_XML "<dds><data_reader><topic><kind>NO_KEY</kind><name>%s</name><dataType>%s</dataType></topic></data_reader></dds>"
#define UXR_TOPIC_XML "<dds><topic><name>%s</name><dataType>%s</dataType></topic></dds>"

pablogs9, 2021-01-25:
Try to use the [same XML that micro-ROS uses](https://github.com/micro-ROS/rmw-microxrcedds/blob/f604270e89a3353eecfc78e7c592d163125ad3a4/rmw_microxrcedds_c/src/utils.c#L302):

```xml
<dds>
   <data_writer>
      <historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
      <qos>
         <reliability>
            <kind>BEST_EFFORT</kind>
         </reliability>
      </qos>
      <topic>
         <kind>NO_KEY</kind>
         <name>[NAME]</name>
         <dataType>[TYPE]</dataType>
         <historyQos>
            <kind>KEEP_ALL</kind>
         </historyQos>
      </topic>
   </data_writer>
</dds>
```

eranroll, 2021-01-25:
Thanks! I will give it a try.

What are the options instead of KEEP_ALL?

pablogs9, 2021-01-25:
You can find all the documentation in the XML profiles of FastDDS. For example this one is [here](https://fast-dds.docs.eprosima.com/en/latest/fastdds/xml_configuration/common.html?highlight=historyQos#historyqos).

eranroll, 2021-01-25:
Everything seems to work as expected now. Thank you very much for your help!

Btw, "durability" should be added to the datawriter / datareader QoS? Or is the history "keep last" attribute in the topic enough?

pablogs9, 2021-01-25:
I'm not sure right now, I guess that they are different configurations (that of course have defaults values), you can find more information in the FastDDS documentation, and do not hesitate to open an issue in any of their repos if you have any doubt with DDS QoS.

Thanks a lot!

Closing bc solved.

closed 2021-01-25 by pablogs9