eProsima/Micro-XRCE-DDS-Client #186: How to add transport descriptor in xml when create a participant

liyue75, 2021-02-10:
When I create a participant using **xur_buffer_create_participant_xml**
and my participant_xml is 
const char* participant = 

                                                   "<dds>"  
                                                    "<transport_descriptors>" 
                                                        "<transport_descriptor>" 
                                                              "<transport_id>"udp</transport_id>"
                                                              "<type>UDPv4</type>"
                                                       "</transport_descriptor>"
                                                    "<transport_descriptors>"
                                                    "<participant>"
                                                        "<domainId>231</domainId>"
                                                     "<rtps>"
                                                           "<name>default_xrce_participant</name>"
                                                           "<useBuiltinTransports>false</useBuiltinTransports>"
                                                            "<userTransports>"
                                                                  "<transport_id>"udp</transport_id>"
                                                            "</userTransports>"
                                                       "</rtps>"     
                                                      "</participant>"
                                                     "</dds>"   

and then use
`uint16_t participant_req = uxr_buffer_create_participant_xml(&session, best_effort_out, participant_id, 0, participant_xml, UXR_REPLACE);`
to add this string to buffer,
But after I send this to the agent, the xrce-agent doesn't seem to able to parse this string, and it returns the following error
`[XMLPARSER Error] Not expected tags: 'transport_descriptors' -> Function parse XML`

So how to add transport descriptor when create a participant in XRCE-DDS?

pablogs9, 2021-02-10:
Hello @liyue75, In the XML passed to the XRCE library, you should keep only the `<participant>` but you can't set the `<transport_descriptors>` inside the participant.

In any case, why do you want to set the `transport_descriptor` to UDPv4? As far as I know, this is the default, isn't it @jamoralp?

jamoralp, 2021-02-10:
Good morning @liyue75 I think the reason this might not be working is becaused you missed a slash `/` when closing the `<transport_descriptors>` tag in line 7. The XML parser of Fast-DDS should be able to recognize this configuration, according to the documentation of Fast-DDS 2.0.0 (link [here](https://fast-rtps.docs.eprosima.com/en/v2.0.0/fastdds/transport/transport.html#transports)). Anyway, as @pablogs9 stated, the default value is UDPv4 already. Please report if this is working for you

liyue75, 2021-02-10:
Thank you for your reply.
@pablogs9   I use 'transport_descriptors' because the actual transport I want to use is UDPv6, and I test with UDPv4 first to see whether the descriptor works.
@jamoralp The XML I actually used has / before transport_descriptors,  If / is missed, the error returned from agent is `[XMLPARSER Error] Error parsing XML buffer -> Function loadXML`

If I can not set `<transport_descriptors>` inside participant, is there any other way to set `transport_descriptors`?

pablogs9, 2021-02-10:
Have you tried to use XRCE Client by references and set your transport in the reference file?

liyue75, 2021-02-10:
@pablogs9 No, I haven't used it and I don't know how to use it yet. Is there any documentation for using it?

pablogs9, 2021-02-10:
Of course: https://micro-ros.github.io/docs/tutorials/core/create_dds_entities_by_ref/

pablogs9, 2021-02-10:
We found a solution, test this XML:

```xml
<dds>
	<profiles>
		<transport_descriptors>
			<transport_descriptor>
				<transport_id>udp</transport_id>
				<type>UDPv4</type>
			</transport_descriptor>
		</transport_descriptors>
	</profiles>
	<participant>
                <domainId>231</domainId>
		<rtps>
			<name>default_xrce_participant</name>
			<useBuiltinTransports>false</useBuiltinTransports>
			<userTransports>
				<transport_id>udp</transport_id>
			</userTransports>
		</rtps>
	</participant>
</dds>
```

pablogs9, 2021-02-10:
I'm closing, please do not hesitate to reopen if you find any other problem

closed 2021-02-10 by pablogs9

liyue75, 2021-02-11:
Hi, @pablogs9, your method works for adding transport_descriptors.
But there is another problem, I find that the tag <domainId> does not work.
In my previous XML file, I specified domainId 231, but the discovery broadcast port was still 7400.

pablogs9, 2021-02-11:
As far as I know, XRCE messages for creating a participant has a field for domain id: check [here](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/687c3e6182fad952e281d29247e848f27fe1f2b0/include/uxr/client/core/session/create_entities_xml.h#L45). I'm not sure about the behavior of the agent when the XML and this number does not match, any idea @jamoralp?

jamoralp, 2021-02-11:
Good morning, the domainID specified in the XML should be ignored. The valid DomainID variable is the one set by the `uxr_buffer_create_participant_xml` function. It is kind of weird to have this option duplicated, but that is what the standard defines

liyue75, 2021-02-11:
Oh, I see. Thank you.