eProsima/Micro-XRCE-DDS-Client #195: Discovery doesn't work

tieugiang91, 2021-02-19:
Environment:
MicroXRCEAgent: Raspberry Pi 4, Ubuntu 20.10
MicroXRCEClient: Discovery example on laptop, Ubuntu 20.04
Connection: Home wifi, the 2 devices could ping each other normally.
Issue: The discovery code built couldn't discover the Agent running on Raspberry Pi, the Discovery runs for some seconds without giving any outputs. The Agent side shows no connection established.

As far as I understand, the on_agent_found callback should be invoked and show the found agent and port. However, it doesn't seem to work. Please correct me if the understanding is wrong, thank you.

pablogs9, 2021-02-19:
How are you launching the Micro XRCE-DDS Agent? Could you share the command?

tieugiang91, 2021-02-19:
I launched it with the command:
./MicroXRCEAgent
then
udp4 --port 2020
The other clients on ESP8266 could connect to that Agent using fixed IP and port.

pablogs9, 2021-02-19:
To enable discovery you should use `./MicroXRCEAgent upd4 --port 2020 -d`

tieugiang91, 2021-02-19:
Thank you for the support, the client discovered the agent.

closed 2021-02-19 by tieugiang91