eProsima/Micro-XRCE-DDS-Client #196: How to use best-effort stream to receive data?

liyue75, 2021-02-19:
Hi, I want to test the latency between a Micro-DDS-Client and a global FAST-DDS node and meet a problem.
My way to estimate the latency is to caculate the roundtrip.

If I use **reliable stream**, it seems to be working well, and related parts of my code:
```
   ......
    uint8_t output_reliable_stream_buffer[2048];    
    uint8_t input_reliable_stream_buffer[2048];   
    uxrStreamId reliable_out = uxr_create_output_reliable_stream(&session, output_reliable_stream_buffer, 2048, 4);
    uxrStreamId reliable_in =  uxr_create_input_reliable_stream(&session, input_reliable_stream_buffer, 2048, 4);         
/* then I create participant, publisher, subscriber, datawriter, datareader in omitted section below */
    ......
/* request data  */
    uxrDeliverControl delivery_control = {0};
    devery_control.max_samples = UXR_MAX_SAMPLES_UNLIMITED;
    uint16_t read_data_req = uxr_buffer_request_data(&session, reliable_out, datareader_id, reliable_in, &delivery_control);
    uint8_t status[1];
    uint8_t requests[1];
/*  in test loop */
    ......
    uxr_parepare_output_stream(&session, reliable_out, datawriter_id, &ub, topic_size);
    HelloWorld_serialize_topic(&ub, &topic);
    start_time = xtimer_now_usec();
    uxr_run_session_until_all_status(&session, 1000, requests, status, 1);
    ......
```
With the code above, the global Fast-DDS node running on the same host as Micro-XRCE-DDS-Agent can receive the msg from the Micro-XRCE-DDS-Client, and the global Fast-DDS node will pub the message to another topic, and the DDS-Client will receive the msg and calculate the roundtrip time.

The problem is:
The roundtrip time caculated by the code above is about 3ms, and is not stable, sometimes 10ms.
I think the latency is a little long (ping time for same data size is about 1.5ms), so I decide to try best-effort stream.
I modified the code above:
```
    ......
    uint8_t output_best_effort_stream_buffer[2048];    
    uxrStreamId best_effort_out = uxr_create_output_best_effort_stream(&session, output_reliable_stream_buffer, 2048);
    uxrStreamId best_effort_in =  uxr_create_input_best_effort_stream(&session);         
/* then I create participant, publisher, subscriber, datawriter, datareader in omitted section below */
    ......
/* request data  */
    uxrDeliverControl delivery_control = {0};
    devery_control.max_samples = UXR_MAX_SAMPLES_UNLIMITED;
    uint16_t read_data_req = uxr_buffer_request_data(&session, best_effort_out, datareader_id, best_effort_in, &delivery_control);
/*  in test loop */
    ......
    uxr_parepare_output_stream(&session, best_effort_out, datawriter_id, &ub, topic_size);
    HelloWorld_serialize_topic(&ub, &topic);
    start_time = xtimer_now_usec();
    uxr_flash_output_streams(&session);
    ......
```
I can send data to global Fast-DDS node, but there seems to be no API for accepting data when using best-effort stream.
If I use uxr_run_session_until_all_status() instead of uxr_run_session_until_timeout() to receive data, it will cause error.

So I have two questions:
1. If I use best-effort stream to send and receive data, will round-trip time be reduced?
2. How to accept data when using best-effort stream?

pablogs9, 2021-02-19:
You should be able to receive data in the best effort stream, in the same way, using the on_topic callback. which error are you having in `uxr_set_topic_callback`?

Regarding your questions:
1. I guess that it should be reduced a bit
2. Using `uxr_set_topic_callback` and running the session with `uxr_set_topic_callback`.

liyue75, 2021-02-20:
@pablogs9,  
I've already set up a callback with `uxr_set_topic_callback`, it will be called in **uxr_run_session_...** these functions.
But when using best-effort stream, if I use `uxr_run_session_timeout` or `uxr_run_session_until_all_status` or other **uxr_run_session...** functions to send and receive data, the data can be send, but then the program will report a segment error
or floating point exception.
However these errors do not occur when using reliable stream, so I think maybe I can not use `uxr_run_session` functions to send and receive data when using best-effort stream, 
If I use `uxr_flash_output_streams(&session)` to send data separately, there is no API avaliable to receive data separately.

My question is: When using best-effort stream, can I use `uxr_run_session_until_all_status` to send and receive data? If so, there is something wrong with my program that causes the error.
BTW, I can not find any examples of  sending and receiving data with best-effort stream, all examples use reliable stream.

pablogs9, 2021-02-22:
Hello @liyue75, can you try [this branch](https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/197)? I have added some examples on how to use the best effort XRCE streams to create the same Hello World pub/sub.

Let me know if this solves your issue.

Thanks!

liyue75, 2021-02-24:
@pablogs9 
Thank you for your help. Now I can use best-effort stream to receive data.
I find  why I could not use best-effort stream to receive data before:
When I used best-effort stream, I only created 2 streams: **best_effort_out** and **best_effort_in**.
After I  looked at your example, I created 3 streams: **best_effort_out**, **besteffort_in** and **reliable_in**, then run the program and the error did not appear.
So the solution is to create a **reliable_in**stream.
Why I have to create a reliable-in stream when using best-effort-stream to send and recv data, is it related to heartbeat?

And the latency decrease  when using best_effort stream to send and receive data compared to reliable stream on STM32 board running a RIOT OS. 
On stm32, when using reliable stream, the latency is 4ms~10ms, when using best-effort stream , the latency is 2.2ms~4ms. 
But if I run the program on computer, the latency when using reliable stream is  0.6~0.8ms. 
I don't konw why the latency on stm32 is unstable, doesit has anything to do with heartbeat? If I increase the heartbeat interval or cancel the heartbeat transmission, will the latency be stable?( The way I port xrce-dds to riot is to modify udp_transport_posix_nopoll.c by riot's posix like socket functions)

pablogs9, 2021-02-24:
You need to create the [reliable out](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/e943a674d6e09fdd2d25d469df2c57c81ddd5158/examples/PublishHelloWorldBestEffort/main.c#L59) and the [reliable in](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/e943a674d6e09fdd2d25d469df2c57c81ddd5158/examples/PublishHelloWorldBestEffort/main.c#L62) because entities are being created reliably, check that all `uxr_buffer_create_*_xml` are using the `reliable_out` stream to communicate with the agent, in fact when creating entities we are [running the session until all status](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/e943a674d6e09fdd2d25d469df2c57c81ddd5158/examples/PublishHelloWorldBestEffort/main.c#L106) which is a concept that is only valid in a reliable stream (a best-effort one does not confirm status).

So, you need to create reliable in/out streams in order to create the entities reliably, but of course you can also use a best effort stream to do so: just use your best effort out the streams on all the `uxr_buffer_create_*_xml` and replace the `uxr_run_session_until_all_status` with something a call to `uxr_run_session_time`.

Let me know if this solves your doubts or if you need an example of this full best effort example.

liyue75, 2021-02-24:
@pablogs9 
Thank you. This solves my doubts.
Now, I use reliable-stream to create entities and best-effort stream to send and receive data, and the latency now is stable at 2.5ms.

closed 2021-02-24 by liyue75