eProsima/Micro-XRCE-DDS-Client #208: Micro-XRCE-DDS on ESP8266 using PlatformIO + Arduino library

tieugiang91, 2021-03-05:
I'm trying to apply Micro-XRCE-DDS on a NodeMCU board with ESP8266 microprocessor.
The IDE is Visual Studio Code with PlatformIO plugin, Arduino is used for Wifi connection. Micro-XRCE-DDS library is also included. However, there are errors in udp_transport_posix.h at:
#include <sys/socket.h>
#include <poll.h>
It seems that the 2 libraries is on Linux only, not on arduinoespressif8266. Is there any customization or configuration that I can do to make it work on that specific board, please?

pablogs9, 2021-03-05:
Hello @tieugiang91,

Some points:
 - Notice that you also have a [polling free UDP transport](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/develop/include/uxr/client/profile/transport/ip/udp/udp_transport_posix_nopoll.h) have you tried to use it?
 - Are you using a bare XRCE-DDS application for some reason? Have you considered micro-ROS?
 - If you find problems using the default transports, notice that we have in `develop` branch a new feature for creating custom transports, so you only need to write four callbacks and bind them at runtime. You have a preliminary version of the documentation [here](https://github.com/eProsima/Micro-XRCE-DDS-docs/blob/feature/api_updates/docs/transport.rst#custom-transport)

Please try to use the polling-less transports, and if it does not fit your requirements, we can talk here about how to use custom transports.

It would be super nice to have some explanations and steps on how to replicate your use case.

tieugiang91, 2021-03-05:
Thank you for the answer.
- I'm using a bare XRCE-DDS application without micro-ROS since micro-ROS supports ESP32 while mine is NodeMCU ESP8266. I'm not quite sure that would be compatible.
- The use case is not complicating. It's just flashing "hello world" to the NodeMCU board. Steps are listed below:
1. Create a PlatformIO project with Arduino framework and NodeMCU 12-E
2. Copy the folders "ucdr" and "uxr" from "/usr/local/include" to the include folder of the project.
(Linking to the "/usr/local/include" in project configuration also leads to the same result)
3. Copy Hello world source code from example to the "src" folder of the project.
Then the project cannot be built as <sys/socket.h> and <poll.h> cannot be found. Even if I tried to add the link to "/usr/include/sys", it also returns errors kind of "unknown type name size_t", etc.
4. In project configurations:
build_flags =
  -I/usr/local/include
  -L/usr/local/lib
  -lmicrocdr
  -lmicroxrcedds_client
has been added. Still, that doesn't help.

pablogs9, 2021-03-05:
In which point you are building the XRCE-DDS Client library?

tieugiang91, 2021-03-05:
I built it in /home/username/Micro-XRCE-DDS-Client directory.

pablogs9, 2021-03-05:
But how are you building the library?

tieugiang91, 2021-03-05:
I cloned it from Github and followed the instructions in the documentation at: https://micro-xrce-dds.docs.eprosima.com/en/latest/installation.html#installing-the-client-stand-alone

pablogs9, 2021-03-05:
But you are building the XRCE-DDS Client library for your computer, it is not being cross-compiled for ESP8266, are you aware of that?

tieugiang91, 2021-03-07:
Ah, thank you, I've got the point, let me try it then.

pablogs9, 2021-03-09:
Could you please provide some steps for reproducing your scenario and use case?

pablogs9, 2021-03-09:
I guess that you should be cross-compiling the library using CMake and a CMake toolchain. Once you have a static library (.a) you should link your app against it.

tieugiang91, 2021-03-09:
Oh, that's it. Thank you, let me try it then.

tieugiang91, 2021-04-26:
@pablogs9 After using Cmake and CMake toolchain to build the library, I got the error:
set_target_properties called with incorrect number of arguments.
Adding "" as:
set_target_properties(${PROJECT_NAME} PROPERTIES
    VERSION
        "${PROJECT_VERSION}"
    SOVERSION
        "${PROJECT_VERSION_MAJOR}.${PROJECT_VERSION_MINOR}"
    C_STANDARD
        99
    C_STANDARD_REQUIRED
        YES
    POSITION_INDEPENDENT_CODE
	    "${UCLIENT_PIC}"
    )
at line 270 CMakeList.txt fixed the problem but led to the next error:
CMake Error at /usr/share/cmake-3.16/Modules/WriteBasicConfigVersionFile.cmake:43 (message):
  No VERSION specified for WRITE_BASIC_CONFIG_VERSION_FILE()
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/CMakePackageConfigHelpers.cmake:225 (write_basic_config_version_file)
  CMakeLists.txt:461 (write_basic_package_version_file)

Steps to reproduce:
1. Install RTOS ESP8266 SDK
2. Add include($ENV{IDF_PATH}/tools/cmake/project.cmake) to the CMakeList.txt of Micro-XRCE-DDS-Client project
3. Add toolchain file into the toolchains folder.

> include(CMakeForceCompiler)

> set(CMAKE_SYSTEM_NAME Generic)
> set(PLATFORM_NAME "FreeRTOS_Plus_TCP")
> set(CMAKE_CROSSCOMPILING True)

> set(CMAKE_C_COMPILER xtensa-lx106-elf-gcc)
> set(CMAKE_CXX_COMPILER xtensa-lx106-elf-g++)
> set(CMAKE_ASM_COMPILER xtensa-lx106-elf-gcc)

> set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
> set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
> set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
> set(CMAKE_FIND_ROOT_PATH_MODE_PACKAGE ONLY)

4. Run cmake command: cmake -DCMAKE_TOOLCHAIN_FILE=../toolchains/esp8266_toolchain.cmake ..
Things went well.
5. The error occurred after "make" command

May you kindly suggest a solution please?

pablogs9, 2021-04-26:
Can you paste here the error that appears when `make` is invoked?

> Run cmake command: cmake -DCMAKE_TOOLCHAIN_FILE=../toolchains/esp8266_toolchain.cmake ..
> Things went well.

The error pasted in your last comment is a CMake error but you say that the CMake step went well.

tieugiang91, 2021-04-26:
Sorry for the unclear description. The CMake error occured after the "make" command, not the CMake command as images below.
[image]
[image]

pablogs9, 2021-04-26:
Could you please copy-paste that text instead of screenshotting?

tieugiang91, 2021-04-26:
The result of:
```
cmake -DCMAKE_TOOLCHAIN_FILE=../toolchains/esp8266_toolchain.cmake ..
```
is:
```
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- IDF_TARGET not set, using default target: esp8266
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/tyna/esp/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
-- Check for working C compiler: /home/tyna/esp/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
-- Check for working C compiler: /home/tyna/esp/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/tyna/esp/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++
-- Check for working CXX compiler: /home/tyna/esp/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: v2.0.0-dirty
-- Building ESP-IDF components for target esp8266
-- Checking Python dependencies...
Python requirements from /home/tyna/esp/ESP8266_RTOS_SDK/requirements.txt are satisfied.
-- Found PythonInterp: /usr/bin/python (found version "3.8.5") 
-- Found Perl: /usr/bin/perl (found version "5.30.0") 
-- Adding linker script /home/tyna/Data110220/Ubuntu/Makino/DDS/xrcedds/Micro-XRCE-DDS-Client/build/esp-idf/esp8266/esp8266_out.ld
-- Adding linker script /home/tyna/esp/ESP8266_RTOS_SDK/components/esp8266/ld/esp8266.project.ld.in
-- Adding linker script /home/tyna/esp/ESP8266_RTOS_SDK/components/esp8266/ld/esp8266.rom.ld
-- Adding linker script /home/tyna/esp/ESP8266_RTOS_SDK/components/esp8266/ld/esp8266.peripherals.ld
-- Components: app_update bootloader bootloader_support coap console esp-tls esp-wolfssl esp8266 esp_common esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota esp_ringbuf esptool_py fatfs freemodbus freertos heap http_parser jsmn json libsodium log lwip mbedtls mdns mqtt newlib nvs_flash openssl partition_table protobuf-c protocomm pthread spi_flash spi_ram spiffs tcp_transport tcpip_adapter vfs wear_levelling wifi_provisioning wpa_supplicant
-- Component paths: /home/tyna/esp/ESP8266_RTOS_SDK/components/app_update /home/tyna/esp/ESP8266_RTOS_SDK/components/bootloader /home/tyna/esp/ESP8266_RTOS_SDK/components/bootloader_support /home/tyna/esp/ESP8266_RTOS_SDK/components/coap /home/tyna/esp/ESP8266_RTOS_SDK/components/console /home/tyna/esp/ESP8266_RTOS_SDK/components/esp-tls /home/tyna/esp/ESP8266_RTOS_SDK/components/esp-wolfssl /home/tyna/esp/ESP8266_RTOS_SDK/components/esp8266 /home/tyna/esp/ESP8266_RTOS_SDK/components/esp_common /home/tyna/esp/ESP8266_RTOS_SDK/components/esp_event /home/tyna/esp/ESP8266_RTOS_SDK/components/esp_gdbstub /home/tyna/esp/ESP8266_RTOS_SDK/components/esp_http_client /home/tyna/esp/ESP8266_RTOS_SDK/components/esp_http_server /home/tyna/esp/ESP8266_RTOS_SDK/components/esp_https_ota /home/tyna/esp/ESP8266_RTOS_SDK/components/esp_ringbuf /home/tyna/esp/ESP8266_RTOS_SDK/components/esptool_py /home/tyna/esp/ESP8266_RTOS_SDK/components/fatfs /home/tyna/esp/ESP8266_RTOS_SDK/components/freemodbus /home/tyna/esp/ESP8266_RTOS_SDK/components/freertos /home/tyna/esp/ESP8266_RTOS_SDK/components/heap /home/tyna/esp/ESP8266_RTOS_SDK/components/http_parser /home/tyna/esp/ESP8266_RTOS_SDK/components/jsmn /home/tyna/esp/ESP8266_RTOS_SDK/components/json /home/tyna/esp/ESP8266_RTOS_SDK/components/libsodium /home/tyna/esp/ESP8266_RTOS_SDK/components/log /home/tyna/esp/ESP8266_RTOS_SDK/components/lwip /home/tyna/esp/ESP8266_RTOS_SDK/components/mbedtls /home/tyna/esp/ESP8266_RTOS_SDK/components/mdns /home/tyna/esp/ESP8266_RTOS_SDK/components/mqtt /home/tyna/esp/ESP8266_RTOS_SDK/components/newlib /home/tyna/esp/ESP8266_RTOS_SDK/components/nvs_flash /home/tyna/esp/ESP8266_RTOS_SDK/components/openssl /home/tyna/esp/ESP8266_RTOS_SDK/components/partition_table /home/tyna/esp/ESP8266_RTOS_SDK/components/protobuf-c /home/tyna/esp/ESP8266_RTOS_SDK/components/protocomm /home/tyna/esp/ESP8266_RTOS_SDK/components/pthread /home/tyna/esp/ESP8266_RTOS_SDK/components/spi_flash /home/tyna/esp/ESP8266_RTOS_SDK/components/spi_ram /home/tyna/esp/ESP8266_RTOS_SDK/components/spiffs /home/tyna/esp/ESP8266_RTOS_SDK/components/tcp_transport /home/tyna/esp/ESP8266_RTOS_SDK/components/tcpip_adapter /home/tyna/esp/ESP8266_RTOS_SDK/components/vfs /home/tyna/esp/ESP8266_RTOS_SDK/components/wear_levelling /home/tyna/esp/ESP8266_RTOS_SDK/components/wifi_provisioning /home/tyna/esp/ESP8266_RTOS_SDK/components/wpa_supplicant
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tyna/Data110220/Ubuntu/Makino/DDS/xrcedds/Micro-XRCE-DDS-Client/build
```
And the result of
```
make
```
is:
```
Scanning dependencies of target uclient
[  0%] Creating directories for 'uclient'
[  0%] No download step for 'uclient'
[  0%] No patch step for 'uclient'
[  1%] No update step for 'uclient'
[  1%] Performing configure step for 'uclient'
loading initial cache file /home/tyna/Data110220/Ubuntu/Makino/DDS/xrcedds/Micro-XRCE-DDS-Client/build/uclient-prefix/tmp/uclient-cache-.cmake
-- Project version: v2.0.0-dirty
-- Building ESP-IDF components for target esp8266
-- Checking Python dependencies...
Python requirements from /home/tyna/esp/ESP8266_RTOS_SDK/requirements.txt are satisfied.
[... cut ...]
  /usr/share/cmake-3.16/Modules/CMakePackageConfigHelpers.cmake:225 (write_basic_config_version_file)
  CMakeLists.txt:461 (write_basic_package_version_file)

-- Configuring incomplete, errors occurred!
See also "/home/tyna/Data110220/Ubuntu/Makino/DDS/xrcedds/Micro-XRCE-DDS-Client/build/CMakeFiles/CMakeOutput.log".
make[2]: *** [CMakeFiles/uclient.dir/build.make:107: uclient-prefix/src/uclient-stamp/uclient-configure] Error 1
make[1]: *** [CMakeFiles/Makefile2:1165: CMakeFiles/uclient.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

pablogs9, 2021-04-26:
Try with `cmake -DCMAKE_TOOLCHAIN_FILE=../toolchains/esp8266_toolchain.cmake -DUCLIENT_SUPERBUILD=OFF ..`

pablogs9, 2021-04-26:
You will need to build separately Micro CDR and Micro XRCE-DDS Client and probably use the `include` folder of the CDR for the XRCE-DDS Client build.

tieugiang91, 2021-04-26:
```
cmake -DCMAKE_TOOLCHAIN_FILE=../toolchains/esp8266_toolchain.cmake -DUCLIENT_SUPERBUILD=OFF ..
```
gives the same error.

pablogs9, 2021-04-26:
I have done this:

```bash
mkdir my_project && cd my_project
export ROOT=$(pwd)

wget https://dl.espressif.com/dl/xtensa-lx106-elf-gcc8_4_0-esp-2020r3-linux-amd64.tar.gz # From here: https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/get-started/linux-setup.html
tar -xzf xtensa-lx106-elf-gcc8_4_0-esp-2020r3-linux-amd64.tar.gz

cd $ROOT
git clone -b v1.2.1 https://github.com/eProsima/Micro-CDR
cd Micro-CDR
mkdir build && cd build
cmake -DCMAKE_TOOLCHAIN_FILE=$ROOT/toolchain.cmake -DCMAKE_INSTALL_PREFIX=$ROOT/install ..
make
make install

cd $ROOT
git clone -b master https://github.com/eProsima/Micro-XRCE-DDS-Client
cd Micro-XRCE-DDS-Client
mkdir build && cd build
cmake -DCMAKE_TOOLCHAIN_FILE=$ROOT/toolchain.cmake -DUCLIENT_SUPERBUILD=OFF -DCMAKE_PREFIX_PATH=$ROOT/install -DCMAKE_INSTALL_PREFIX=$ROOT/install .. 
make 
make install
```
This should give an error because of FreeRTOS includes that should be added to the toolchain.

The toolchain.cmake is:

```cmake
include(CMakeForceCompiler)
set(CMAKE_SYSTEM_NAME Generic)
set(PLATFORM_NAME "FreeRTOS_Plus_TCP")
set(CMAKE_CROSSCOMPILING True)

SET (CMAKE_C_COMPILER_WORKS 1)
SET (CMAKE_CXX_COMPILER_WORKS 1)

set(CMAKE_C_COMPILER $ENV{ROOT}/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc)
set(CMAKE_CXX_COMPILER $ENV{ROOT}/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++)
set(CMAKE_ASM_COMPILER $ENV{ROOT}/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc)

set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_PACKAGE ONLY)
include_directories(SYSTEM 
// Fill here your FreeRTOS includes
)
```

tieugiang91, 2021-04-27:
Thank you for the reply. The problem is I also don't know the correct way to include ESP8266 RTOS SDK library without including the default project.cmake as in the hello-world template at 
https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/examples/get-started/hello_world
It seems that FreeRTOS should be cross-compiled again for the NodeMCU board, doesn't it?

tieugiang91, 2021-05-10:
After manually including FreeRTOS libraries and headers and following the steps above, I still got the following errors while buiding with CMake:
```
Scanning dependencies of target microxrcedds_client
[  3%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/stream/input_best_effort_stream.c.obj
[  6%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/stream/input_reliable_stream.c.obj
[  9%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/stream/output_best_effort_stream.c.obj
[ 12%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/stream/output_reliable_stream.c.obj
[ 15%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/stream/stream_storage.c.obj
[ 18%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/stream/stream_id.c.obj
[ 21%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/stream/seq_num.c.obj
[ 25%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/session.c.obj
[ 28%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/session_info.c.obj
[ 31%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/submessage.c.obj
[ 34%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/object_id.c.obj
[ 37%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/serialization/xrce_types.c.obj
[ 40%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/serialization/xrce_header.c.obj
[ 43%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/serialization/xrce_subheader.c.obj
[ 46%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/util/time.c.obj
[ 50%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/util/ping.c.obj
In file included from /home/tyna/Data110220/Ubuntu/Makino/DDS/xrcedds/Micro-XRCE-DDS-Client/include/uxr/client/transport.h:43,
                 from /home/tyna/Data110220/Ubuntu/Makino/DDS/xrcedds/Micro-XRCE-DDS-Client/include/uxr/client/util/ping.h:24,
                 from /home/tyna/Data110220/Ubuntu/Makino/DDS/xrcedds/Micro-XRCE-DDS-Client/src/c/util/ping.c:1:
/home/tyna/Data110220/Ubuntu/Makino/DDS/xrcedds/Micro-XRCE-DDS-Client/include/uxr/client/profile/transport/ip/tcp/tcp_transport.h:51:27: error: field 'platform' has incomplete type
     struct uxrTCPPlatform platform;
                           ^~~~~~~~
In file included from /home/tyna/Data110220/Ubuntu/Makino/DDS/xrcedds/Micro-XRCE-DDS-Client/include/uxr/client/transport.h:52,
                 from /home/tyna/Data110220/Ubuntu/Makino/DDS/xrcedds/Micro-XRCE-DDS-Client/include/uxr/client/util/ping.h:24,
                 from /home/tyna/Data110220/Ubuntu/Makino/DDS/xrcedds/Micro-XRCE-DDS-Client/src/c/util/ping.c:1:
/home/tyna/Data110220/Ubuntu/Makino/DDS/xrcedds/Micro-XRCE-DDS-Client/include/uxr/client/profile/transport/serial/serial_transport.h:35:30: error: field 'platform' has incomplete type
     struct uxrSerialPlatform platform;
                              ^~~~~~~~
make[2]: *** [CMakeFiles/microxrcedds_client.dir/build.make:258: CMakeFiles/microxrcedds_client.dir/src/c/util/ping.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/microxrcedds_client.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```
The toolchain file is:
```
include(CMakeForceCompiler)

set(CMAKE_SYSTEM_NAME Generic)
set(PLATFORM_NAME "FreeRTOS_Plus_TCP")
set(CMAKE_CROSSCOMPILING True)

set(CMAKE_C_COMPILER xtensa-lx106-elf-gcc)
set(CMAKE_CXX_COMPILER xtensa-lx106-elf-g++)
set(CMAKE_ASM_COMPILER xtensa-lx106-elf-gcc)

set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_PACKAGE ONLY)

set(FREE_RTOS_DIR $ENV{IDF_PATH}/components/freertos)
set(ESP8266_DIR $ENV{IDF_PATH}/components/esp8266)
include_directories(SYSTEM
	${FREE_RTOS_DIR}/include
	${FREE_RTOS_DIR}/include/freertos
	${FREE_RTOS_DIR}/include/freertos/private
	${FREE_RTOS_DIR}/port/esp8266/include
	${FREE_RTOS_DIR}/port/esp8266/include/freertos
	$ENV{IDF_PATH}/components/heap/include
	$ENV{IDF_PATH}/components/heap/port/esp8266/include
	${ESP8266_DIR}/include
	$ENV{IDF_PATH}/components/esp_common/include
	$ENV{HOME}/Data110220/Ubuntu/Makino/ESP8266/FreeRTOSv202012.00/FreeRTOS-Plus/Source/FreeRTOS-Plus-TCP/include
	$ENV{HOME}/Data110220/Ubuntu/Makino/ESP8266/FreeRTOSv202012.00/FreeRTOS-Plus/Source/FreeRTOS-Plus-TCP/portable/Compiler/GCC
)
```
May you point out what's wrong with the compilation please?

pablogs9, 2021-05-10:
Have you disabled TCP and serial profiles if you are not going to use them?

https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/69134a9ba97f98f3cfad1588af816a6a22133116/CMakeLists.txt#L47

pablogs9, 2021-05-10:
You need to include the header where those objects are defined in your FreeRTOS installation.

tieugiang91, 2021-05-11:
After including the headers, it still gave the following errors:
```
Scanning dependencies of target microxrcedds_client
[  3%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/stream/input_best_effort_stream.c.obj
[  6%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/stream/input_reliable_stream.c.obj
[ 10%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/stream/output_best_effort_stream.c.obj
[ 13%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/stream/output_reliable_stream.c.obj
[ 17%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/stream/stream_storage.c.obj
[ 20%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/stream/stream_id.c.obj
[ 24%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/stream/seq_num.c.obj
[ 27%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/session.c.obj
[ 31%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/session_info.c.obj
[ 34%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/submessage.c.obj
[ 37%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/object_id.c.obj
[ 41%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/serialization/xrce_types.c.obj
[ 44%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/serialization/xrce_header.c.obj
[ 48%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/serialization/xrce_subheader.c.obj
[ 51%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/util/time.c.obj
[ 55%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/util/ping.c.obj
[ 58%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/common_create_entities.c.obj
[ 62%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/create_entities_ref.c.obj
[ 65%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/create_entities_xml.c.obj
[ 68%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/read_access.c.obj
[ 72%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/write_access.c.obj
[ 75%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/profile/transport/stream_framing/stream_framing_protocol.c.obj
[ 79%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/profile/transport/ip/udp/udp_transport.c.obj
[ 82%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/profile/transport/ip/udp/udp_transport_freertos_plus_tcp.c.obj
[ 86%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/profile/transport/ip/ip_freertos_plus_tcp.c.obj
[ 89%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/profile/discovery/discovery.c.obj
[ 93%] Building C object CMakeFiles/microxrcedds_client.dir/src/c/profile/discovery/transport/udp_transport_datagram_freertos_plus_tcp.c.obj
/home/tyna/DDS/Micro-XRCE-DDS-Client/src/c/profile/discovery/transport/udp_transport_datagram_freertos_plus_tcp.c: In function 'uxr_init_udp_transport_datagram':
/home/tyna/DDS/Micro-XRCE-DDS-Client/src/c/profile/discovery/transport/udp_transport_datagram_freertos_plus_tcp.c:29:14: error: 'uxrUDPTransportDatagram' {aka 'struct uxrUDPTransportDatagram'} has no member named 'fd'
     transport->fd = FreeRTOS_socket(FREERTOS_AF_INET, FREERTOS_SOCK_DGRAM, FREERTOS_IPPROTO_UDP);
              ^~
/home/tyna/DDS/Micro-XRCE-DDS-Client/src/c/profile/discovery/transport/udp_transport_datagram_freertos_plus_tcp.c:31:45: error: 'uxrUDPTransportDatagram' {aka 'struct uxrUDPTransportDatagram'} has no member named 'fd'
     if (FREERTOS_INVALID_SOCKET != transport->fd)
                                             ^~
/home/tyna/DDS/Micro-XRCE-DDS-Client/src/c/profile/discovery/transport/udp_transport_datagram_freertos_plus_tcp.c:33:18: error: 'uxrUDPTransportDatagram' {aka 'struct uxrUDPTransportDatagram'} has no member named 'poll_fd'
         transport->poll_fd = FreeRTOS_CreateSocketSet();
                  ^~
/home/tyna/DDS/Micro-XRCE-DDS-Client/src/c/profile/discovery/transport/udp_transport_datagram_freertos_plus_tcp.c:35:30: error: 'uxrUDPTransportDatagram' {aka 'struct uxrUDPTransportDatagram'} has no member named 'poll_fd'
         if (NULL != transport->poll_fd)
                              ^~
/home/tyna/DDS/Micro-XRCE-DDS-Client/src/c/profile/discovery/transport/udp_transport_datagram_freertos_plus_tcp.c:38:38: error: 'uxrUDPTransportDatagram' {aka 'struct uxrUDPTransportDatagram'} has no member named 'fd'
             FreeRTOS_FD_SET(transport->fd, transport->poll_fd, eSELECT_READ);
                                      ^~
/home/tyna/DDS/Micro-XRCE-DDS-Client/src/c/profile/discovery/transport/udp_transport_datagram_freertos_plus_tcp.c:38:53: error: 'uxrUDPTransportDatagram' {aka 'struct uxrUDPTransportDatagram'} has no member named 'poll_fd'
             FreeRTOS_FD_SET(transport->fd, transport->poll_fd, eSELECT_READ);
                                                     ^~
/home/tyna/DDS/Micro-XRCE-DDS-Client/src/c/profile/discovery/transport/udp_transport_datagram_freertos_plus_tcp.c: In function 'uxr_close_udp_transport_datagram':
/home/tyna/DDS/Micro-XRCE-DDS-Client/src/c/profile/discovery/transport/udp_transport_datagram_freertos_plus_tcp.c:50:42: error: 'uxrUDPTransportDatagram' {aka 'struct uxrUDPTransportDatagram'} has no member named 'fd'
[... cut ...]
#define UXR_CONFIG_MIN_SESSION_CONNECTION_INTERVAL    1000
#define UXR_CONFIG_MIN_HEARTBEAT_TIME_INTERVAL        1

#ifdef UCLIENT_PROFILE_UDP
#define UXR_CONFIG_UDP_TRANSPORT_MTU                  512
#endif
#ifdef UCLIENT_PROFILE_TCP
#define UXR_CONFIG_TCP_TRANSPORT_MTU                  512
#endif
#ifdef UCLIENT_PROFILE_SERIAL
#define UXR_CONFIG_SERIAL_TRANSPORT_MTU               512
#endif
#ifdef UCLIENT_PROFILE_CUSTOM_TRANSPORT
#define UXR_CONFIG_CUSTOM_TRANSPORT_MTU                  512
#endif

#endif // _UXR_CLIENT_CONFIG_H_

```

pablogs9, 2021-05-11:
https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/239 merged, try it now

tieugiang91, 2021-05-11:
Thank you, the fix works well. The rest seems again my header trouble errors, I'll try to fix it then update later.

tieugiang91, 2021-05-17:
Thank you for the support. Finally, the libmicroxrcedds_client.a has been built.
I leave the CMakeList.txt of the example project SubscribeHelloWorld unchanged and compile with:
```
cmake -DCMAKE_TOOLCHAIN_FILE=$TOOLCHAIN_PATH/esp8266_toolchain.cmake -DCMAKE_PREFIX_PATH=$ROOT/install -DCMAKE_INSTALL_PREFIX=$ROOT/install -DUCLIENT_PROFILE_UDP=ON ..
```
then it gave the linking errors after the "make" command:
```
Scanning dependencies of target SubscribeHelloWorldClient
[ 33%] Building C object CMakeFiles/SubscribeHelloWorldClient.dir/main.c.obj
[ 66%] Building C object CMakeFiles/SubscribeHelloWorldClient.dir/HelloWorld.c.obj
[100%] Linking C executable SubscribeHelloWorldClient
/home/tyna/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/tyna/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-stdio.o):(.literal.__sclose+0x0): warning: _close_r is not implemented and will always fail
/home/tyna/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/tyna/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-makebuf.o):(.literal.__swhatbuf_r+0x8): warning: _fstat_r is not implemented and will always fail
/home/tyna/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/tyna/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-stdio.o):(.literal.__swrite+0x4): warning: _lseek_r is not implemented and will always fail
/home/tyna/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/tyna/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-stdio.o):(.literal.__sread+0x4): warning: _read_r is not implemented and will always fail
/home/tyna/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/tyna/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-stdio.o):(.literal.__swrite+0x8): warning: _write_r is not implemented and will always fail
/home/tyna/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/tyna/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-puts.o):(.literal._puts_r+0x24): warning: pthread_setcancelstate is not implemented and will always fail
/home/tyna/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/tyna/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-printf.o):(.literal.printf+0x0): warning: __getreent is not implemented and will always fail
/home/tyna/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/tyna/DDS/install/lib/libmicroxrcedds_client.a(time.c.obj):(.literal.uxr_nanos+0x0): undefined reference to `vTaskSetTimeOutState'
/home/tyna/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/tyna/DDS/install/lib/libmicroxrcedds_client.a(udp_transport_freertos_plus_tcp.c.obj):(.literal.uxr_init_udp_platform+0x4): undefined reference to `FreeRTOS_socket'
/home/tyna/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/tyna/DDS/install/lib/libmicroxrcedds_client.a(udp_transport_freertos_plus_tcp.c.obj):(.literal.uxr_init_udp_platform+0x8): undefined reference to `FreeRTOS_inet_addr'
/home/tyna/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/tyna/DDS/install/lib/libmicroxrcedds_client.a(udp_transport_freertos_plus_tcp.c.obj):(.literal.uxr_init_udp_platform+0xc): undefined reference to `FreeRTOS_CreateSocketSet'
/home/tyna/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/tyna/DDS/install/lib/libmicroxrcedds_client.a(udp_transport_freertos_plus_tcp.c.obj):(.literal.uxr_init_udp_platform+0x10): undefined reference to `FreeRTOS_FD_SET'
/home/tyna/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/tyna/DDS/install/lib/libmicroxrcedds_client.a(udp_transport_freertos_plus_tcp.c.obj):(.literal.uxr_close_udp_platform+0x0): undefined reference to `FreeRTOS_closesocket'
/home/tyna/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/tyna/DDS/install/lib/libmicroxrcedds_client.a(udp_transport_freertos_plus_tcp.c.obj):(.literal.uxr_write_udp_data_platform+0x0): undefined reference to `FreeRTOS_sendto'
/home/tyna/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/tyna/DDS/install/lib/libmicroxrcedds_client.a(udp_transport_freertos_plus_tcp.c.obj):(.literal.uxr_read_udp_data_platform+0x4): undefined reference to `FreeRTOS_select'
/home/tyna/esp/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/8.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/tyna/DDS/install/lib/libmicroxrcedds_client.a(udp_transport_freertos_plus_tcp.c.obj):(.literal.uxr_read_udp_data_platform+0x8): undefined reference to `FreeRTOS_recvfrom'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/SubscribeHelloWorldClient.dir/build.make:101: SubscribeHelloWorldClient] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/SubscribeHelloWorldClient.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```
Are there any wrong configurations, please?

pablogs9, 2021-05-18:
You are not linking against your FreeRTOS libraries, I guess that this is not related to micro-ROS