eProsima/Micro-XRCE-DDS-Client #217: Issue while compiling the client code

arshPratap, 2021-03-26:
I have been following the steps listed in the tutorial about building the client code and I seem to run into the following error.
CMake Error at CMakeLists.txt:106 (find_package):
  By not providing "Findmicrocdr.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "microcdr",
  but CMake did not find one.

  Could not find a package configuration file provided by "microcdr"
  (requested version 1.2.0) with any of the following names:

    microcdrConfig.cmake
    microcdr-config.cmake

  Add the installation prefix of "microcdr" to CMAKE_PREFIX_PATH or set
  "microcdr_DIR" to a directory containing one of the above files.  If
  "microcdr" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/arsh/Micro-XRCE-DDS-Client/build/CMakeFiles/CMakeOutput.log".
Makefile:160: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1

The error is exactly similar to the one that was asked [here](https://github.com/eProsima/Micro-XRCE-DDS-Client/issues/160).
Even after following the intended work around I still seem to have the same problem.
It would be great if I could get some help here.

pablogs9, 2021-03-29:
Hello, I just have followed the instructions and everything works ok:

```bash
git clone -b develop https://github.com/eProsima/Micro-XRCE-DDS-Client
cd Micro-XRCE-DDS-Client
mkdir build && cd build
cmake .. -DUCLIENT_BUILD_EXAMPLES=ON
make
```

Here you have a screencast of my testing session:

https://asciinema.org/a/sV52py9wwGDlwSfJFaiYmHAkl

Let us know if following these instructions everything works.

DenvE, 2021-05-25:
Hi, this issue occurs when following https://github.com/eProsima/Micro-XRCE-DDS-Apps/tree/microxrce_dev/Zephyr, 
clone the git, install zephyr and try to build,..

CMake Error at CMakeLists.txt:110 (find_package):
  Could not find a configuration file for package "microcdr" that exactly
  matches requested version "1.2.0".

  The following configuration files were considered but not accepted:

Rico-Bubble, 2021-11-24:
using the linux default gcc compiler is fine ,but the smae issue occurs when switching the arm toolchain. find lib seems to have the problem,and attempt to manually add lib links did not work. The following is the error message, I will be appreciate that provide idea to fix it
-----------------
-- Install configuration: "Release"
-- Up-to-date: /home/next/Workspace/Micro-XRCE-DDS-Client/build/temp_install/microcdr-1.2.1/lib/libmicrocdr.a
-- Up-to-date: /home/next/Workspace/Micro-XRCE-DDS-Client/build/temp_install/microcdr-1.2.1/include/ucdr
-- Up-to-date: /home/next/Workspace/Micro-XRCE-DDS-Client/build/temp_install/microcdr-1.2.1/include/ucdr/visibility.h
-- Up-to-date: /home/next/Workspace/Micro-XRCE-DDS-Client/build/temp_install/microcdr-1.2.1/include/ucdr/microcdr.h
-- Up-to-date: /home/next/Workspace/Micro-XRCE-DDS-Client/build/temp_install/microcdr-1.2.1/include/ucdr/config.h
-- Up-to-date: /home/next/Workspace/Micro-XRCE-DDS-Client/build/temp_install/microcdr-1.2.1/share/microcdr/cmake/microcdrTargets.cmake
-- Installing: /home/next/Workspace/Micro-XRCE-DDS-Client/build/temp_install/microcdr-1.2.1/share/microcdr/cmake/microcdrTargets-release.cmake
-- Up-to-date: /home/next/Workspace/Micro-XRCE-DDS-Client/build/temp_install/microcdr-1.2.1/share/microcdr/cmake/microcdrConfig.cmake
-- Up-to-date: /home/next/Workspace/Micro-XRCE-DDS-Client/build/temp_install/microcdr-1.2.1/share/microcdr/cmake/microcdrConfigVersion.cmake
[ 31%] Completed 'microcdr'
[ 50%] Built target microcdr
Scanning dependencies of target uclient
[ 56%] Performing configure step for 'uclient'
loading initial cache file /home/next/Workspace/Micro-XRCE-DDS-Client/build/uclient-prefix/tmp/uclient-cache-.cmake
-- Setting build type to 'Release' as none was specified.
CMake Error at CMakeLists.txt:122 (find_package):
  By not providing "Findmicrocdr.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "microcdr",
  but CMake did not find one.

  **Could not find a package configuration file provided by "microcdr"
  (requested version 1.2.1) with any of the following names:**

    microcdrConfig.cmake
    microcdr-config.cmake

  Add the installation prefix of "microcdr" to CMAKE_PREFIX_PATH or set
  "microcdr_DIR" to a directory containing one of the above files.  If
  "microcdr" provides a separate development package or SDK, be sure it has
  been installed.
------------------

pablogs9, 2021-11-24:
Could you provide instructions to reproduce using the ARM toolchain?

closed 2021-12-17 by pablogs9

lxc777, 2022-03-14:
Hi @pablogs9 ,I encountered this problem when compiling agents using the NDK 

[image]

git clone -b develop https://github.com/eProsima/Micro-XRCE-DDS-Client
cd Micro-XRCE-DDS-Client
mkdir build && cd build

 C:\Android_SDK\cmake\3.18.1\bin\cmake.exe -DCMAKE_SYSTEM_NAME=Android -DCMAKE_TOOLCHAIN_FILE=C:\Android_SDK\ndk-bundle\build\cmake\android.toolchain.cmake -DANDROID_ABI=x86 -DANDROID_PLATFORM=android-25 -DANDROID_NDK=C:\Android_SDK\ndk-bundle -DCMAKE_MAKE_PROGRAM=C:\Android_SDK\cmake\3.18.1\bin\ninja.exe -DCMAKE_BUILD_TYPE=Debug -DCMAKE_SYSTEM_NAME=Android -DCMAKE_ANDROID_ARCH_ABI=x86 -DANDROID_NATIVE_API_LEVEL=9 -DCMAKE_SYSTEM_VERSION=25 -GNinja -DUCLIENT_BUILD_EXAMPLES=ON -DCMAKE_ANDROID_NDK_TOOLCHAIN_VERSION=clang -DANDROID_TOOLCHAIN=clang -DCMAKE_GENERATOR="Ninja" ..

C:\Android_SDK\cmake\3.18.1\bin\ninja.exe

shivangsgangadia, 2024-12-01:
Hi, I had the same issue pop up when cross compiling for a raspberry-pi 0W using a buildroot generated toolchain.
The issue was:
When previously building, I succeeded and then tried to install the built binaries onto my buildroot target using the cmake install prefix flag. Apparantly the flag was ignored for some reason and the binaries instead got installed on my host system which included the cmake Find files. This was causing the new cross-compile build to look for the binaries/libraries on my host system.
The issue went away  when I manually deleted the installed binaries from `/usr/local/lib`
For reference, the cmake toolchain file I am using is:
`
set(CMAKE_SYSTEM_NAME Linux)
set(CMAKE_TRY_COMPILE_TARGET_TYPE STATIC_LIBRARY)
set(CMAKE_SYSTEM_VERSION 1)

list(APPEND CMAKE_MODULE_PATH ${CMAKE_CURRENT_LIST_DIR})

set(CMAKE_C_FLAGS_DEBUG "" CACHE STRING "Debug CFLAGS")
set(CMAKE_C_FLAGS_RELEASE " -DNDEBUG" CACHE STRING "Release CFLAGS")

# Build type from the Buildroot configuration
set(USE_NO_MAYBE_UNINITIALIZED 1)
set(USE_NO_UNINITIALIZED 1)

set(CMAKE_C_FLAGS "-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g0 -D_FORTIFY_SOURCE=1" CACHE STRING "Buildroot CFLAGS")
set(CMAKE_EXE_LINKER_FLAGS "" CACHE STRING "Buildroot LDFLAGS for executables")
set(CMAKE_SHARED_LINKER_FLAGS "" CACHE STRING "Buildroot LDFLAGS for shared libraries")
set(CMAKE_MODULE_LINKER_FLAGS "" CACHE STRING "Buildroot LDFLAGS for module libraries")

set(CMAKE_SYSROOT "$ENV{TOOLCHAIN_DIR}/arm-buildroot-linux-gnueabihf/sysroot")
set(CMAKE_FIND_ROOT_PATH ${CMAKE_SYSROOT})
set(CMAKE_LIBRARY_ARCHITECTURE arm-buildroot-linux-gnueabihf)
set(ENV{PKG_CONFIG_SYSROOT_DIR} ${CMAKE_SYSROOT})

# This toolchain file can be used both inside and outside Buildroot.
set(CMAKE_C_COMPILER "$ENV{TOOLCHAIN_DIR}/bin/arm-buildroot-linux-gnueabihf-gcc")
set(CMAKE_CXX_COMPILER "$ENV{TOOLCHAIN_DIR}/bin/arm-buildroot-linux-gnueabihf-g++")

set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wno-maybe-uninitialized -Wno-unused-parameter -Wno-unused-function -lpthread")

set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
`
I am setting the TOOLCHAIN_DIR to buildroot's `output/host`
Also note the additional linkages to libpthread. I found that the linkage set in the CMakeLists.txt doesn't agree with the cross-compile system and had to edit the following out:
line 369, comment out "find_package(Threads REQUIRED)"

While I haven't attempted to build the client libraries myself because I am instead using microROS static libraries built using ROS2, I suspect a similar problem might be causing the missing microcdr error.