eProsima/Micro-XRCE-DDS-Client #235: Multithread includes wrong header FreeRTOS

bjv-capra, 2021-05-07:
I was trying to build the multithread header for FreeRTOS, adding add_definitions(PLATFORM_NAMES_FREERTOS) to my toolchain.
Then I got the compile error
```
Micro-XRCE-DDS-Client/include/uxr/client/profile/multithread/multithread.h:33:10: fatal error: freertos/FreeRTOS.h: No such file or directory
   33 | #include <freertos/FreeRTOS.h>
```

While in other files FreeRTOS is included as
```
#include "FreeRTOS.h"
```

I thought I had built the multithread feature, but I was clearly wrong. This needs to be fixed ASAP

pablogs9, 2021-05-07:
https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/236 merged

closed 2021-05-07 by pablogs9

vtran5, 2023-09-17:
Hi @pablogs9, I would like to reopen this issue as I have the exact same problem when enabling multithreading with FreeRTOS:

`Micro-XRCE-DDS-Client/include/uxr/client/profile/multithread/multithread.h:37:10: fatal error: FreeRTOS.h: No such file or directory
   33 | #include "FreeRTOS.h"`

I'm using FreeRTOS with esp32 and micro_ros_espidf_component Iron release. I am able to use FreeRTOS fine with other components in micro-ros, so not sure what else I need to change.

pablogs9, 2023-09-18:
@vtran5 are you including your FreeRTOS headers during the Micro XRCE-DDS Client CMake build?