eProsima/Micro-XRCE-DDS-Client #242: Disabling IPv6 support

gavanderhoorn, 2021-05-17:
The platform I'm trying to build the client for does not offer support for IPv6.

While it doesn't appear there are many places where `AF_INET6` et al. is referenced, I wanted to ask whether there is a way to disable looking for/using IPv6 at configure time (CMake option perhaps)?

pablogs9, 2021-05-17:
Where is failing your build? Which are the defines or functions in Micro XRCE-DDS Client transports that are not defined in your system? Just `AF_INET6`?

gavanderhoorn, 2021-05-17:
Yes, essentially all files referring to `AF_INET6`, which I believe are these:

```
./src/c/profile/transport/ip/ip_posix.c
./src/c/profile/transport/ip/ip_windows.c
./src/c/profile/transport/ip/tcp/tcp_transport_posix.c
./src/c/profile/transport/ip/tcp/tcp_transport_windows.c
./src/c/profile/transport/ip/udp/udp_transport_posix.c
./src/c/profile/transport/ip/udp/udp_transport_posix_nopoll.c
./src/c/profile/transport/ip/udp/udp_transport_windows.c
```

I could also `#define AF_INET6 ..` to something, but having a configuration option (and accompanying `#ifdef`s perhaps in the code) would communicate intent much better I believe.

pablogs9, 2021-05-17:
Basically, those files are not going to be built if you are in a non-windows platform:

```
./src/c/profile/transport/ip/ip_windows.c
./src/c/profile/transport/ip/tcp/tcp_transport_windows.c
./src/c/profile/transport/ip/udp/udp_transport_windows.c
```

If you are building the library with a CMake cross-compilation toolchain, is possible to add to your toolchain file:

```cmake
add_compile_definitions(AF_INET6=AF_INET)
```

If you are building the library in the platform using a CMake call you can do:

```
cmake .. -DCMAKE_C_FLAGS=-DAF_INET6=AF_INET
```

----

If you want to contribute a PR with an implementation of your proposal, it will be really welcomed. Thanks!

gavanderhoorn, 2021-05-17:
I might contribute a PR, but I'd need some more information/guidance.

Would `#ifdef`-ing the `AF_INET6` related code be acceptable?

pablogs9, 2021-05-17:
Yes, the approach should be adding something like `option(UCLIENT_PROFILE_IPV6 "Enable IPv6 in transports." ON)` and then `ifdef` in every section related to IPv6 in the transports.