eProsima/Micro-XRCE-DDS-Client #249: How to use Micro-XRCE-DDS Client with FreeRTOS

wemaxon, 2021-06-04:
Hello,

I don't know if this is the right place to ask. If it's not, where could I ask instead?

I'd like to use the Micro-XRCE-DDS Client Library on an Ambiq Apollo 3 MCU to communicate with a Raspberry Pi running ROS 2 foxy via UART. I have some experience with programming MCU's, but Real Time OS and using command line tools and makefiles or cmake to compile is completely new to me.

I managed to set up a working FreeRTOS Project with the Ambiq Apollo SDK but i don't know how to employ the Micro-XRCE-DDS Client Library. What would the appropriate workflow be like? Maybe someone could point me in the right direction :)

Should I be running micro-ros on the MCU? If so, why? I would prefer to keep the firmware simple.

Do I have to pre-compile the library? Or could I include the source files in my Makefile?

The example, which I found at https://github.com/eProsima/Micro-XRCE-DDS-Apps/tree/microxrce_dev/FreeRTOS, uses a FreeRTOS version, that implements the POSIX standard. Do I need this version of FreeRTOS?

How is the interface to the UART hardware handled? I expect ill have to implement some handling by myself, where should I do that?

Thanks for your help in advance!

pablogs9, 2021-06-09:
Hello @wemaxon, first of all thanks for the interest in our XRCE-DDS products. The first question is: why are you targeting a bare XRCE-DDS application instead of a micro-ROS application? Is there any reason for that?

wemaxon, 2021-06-10:
Hey @pablogs9, thanks for your reply! I did not now this was the better way to do it. I'll follow the documentation on https://micro.ros.org/ and try to create custom static micro-ROS library and include the library in my project. The micro-ros library will have the Micro-XRCE-DDS Client included, did I understand this correctly?

pablogs9, 2021-06-10:
Yes, the micro-ROS stack has Micro XRCE-DDS as default middleware. Let know you have any issue. I'm closing this, please reopen in a micro-ROS repository if you have any question.

closed 2021-06-10 by pablogs9

pablogs9, 2022-12-21:
Hello @George-Gi, 

micro-ROS are being used in production environments so I guess that if you need to go with it, it is not a bad idea.
You will find all the portability information for micro-ROS here: https://docs.vulcanexus.org/en/latest/rst/tutorials/micro/custom_platforms/custom_platforms.html

In the case of the XRCE-DDS Client, it is just a matter of building the C sources for your platform and being careful about writing the correct transport for your platform: https://micro-xrce-dds.docs.eprosima.com/en/latest/transport.html?highlight=custom%20transport#custom-transport

In any case, do not hesitate to open a new issue in any of our XRCE or micro-ROS repos if you find any problem.
Also if you need professional support, contact us at support@eProsima.com

George-Gi, 2022-12-21:
@pablogs9 thank you for your quick and elaborate response.
I noticed the following section in the micro-ROS github

[image]

and also noticed that in the eProsima website micro-ROS is listed under the R&D products.
So, should I consider it a product ready to be integrated for production purposes?