eProsima/Micro-XRCE-DDS-Client #250: Is there a way to communicate an xrce client to ros2 ?

arshPratap, 2021-06-18:
**Scenario**
I have an idl fil that goes as
``` 
struct Num {
	long num;
};
```
I had created an XRCE client that would connect with the agent to publish the above-mentioned topic.I was able to successfully connect with a DDS sub

Now I wanted to test this experiment with a ros2 node.
I connected the same XRCE client this time to XRCE agent and changed the topic name in accordance with this.Running ros2 node list and topic list have also successfully listed my node name and topic name.Now I had a ros2 node subscribing to the msg equivalent of the above idl file , that is:

```
int32 num
```
Now when I run the sub,the sub is not able to read the topic?Is there a solution to this problem.I am aware of micro ros and eProsima Integration Services ,but I am interested in finding a "native" solution to this problem , something that does not incorporate micro ros and integration services.

**System Information**

1. Ubuntu 20.04
2. XRCE Client v2.0(latest)
3. ROS2 foxy

Thanks !

pablogs9, 2021-06-18:
In general an XRCE-DDS Client participant that is able to communicate with ROS 2 needs:
- A compatible topic and type [**name mangling**](https://design.ros2.org/articles/topic_and_service_names.html), for example, you will need to create a topic named `rt/my_topic` with a type name `std_msgs::dds_::Int32_`
- The CDR serialization should be compatible and it is interesting to check the generated ROS 2 typesupport for Micro XRCE-DDS. If you are using `std_msgs/Int32` the generated code for this type support in micro-ROS is:

```c
static bool _Int32__cdr_serialize(
  const void * untyped_ros_message,
  ucdrBuffer * cdr)
{
  (void) untyped_ros_message;
  (void) cdr;

  bool rv = false;

  if (!untyped_ros_message) {
    fprintf(stderr, "ros message handle is null\n");
    return false;
  }

  _Int32__ros_msg_type * ros_message = (_Int32__ros_msg_type *)(untyped_ros_message);
  (void)ros_message;

  // Member: data
  rv = ucdr_serialize_int32_t(cdr, ros_message->data);

  return rv;
}
```

You should be using something similar.

Can you share your code?

arshPratap, 2021-06-19:
@pablogs9  sure 

here is the link to the repo [https://github.com/arshPratap/ros2client](url)

I have added the ros2 sub code in the link too..

also here are some ss
**(client sending info successfully)**
[image]
**(client connected to agent)**
[image]

**(on the ros2 end,ros2 topic list and ros2 node list detect the client..but sub does not read the message)**

[image]

would be really great if you could look into the code

pablogs9, 2021-06-21:
Share the output of the agent using the flag `-v6`. 
Also, try to listen to the topic with `ros2 topic echo /NumTopic`.

pablogs9, 2021-07-02:
Any update on this?

arshPratap, 2021-07-02:
Hi @pablogs9 so sorry for the delay!.I ran the above mentioned code with the flags and commands you suggested ..attaching the screenshots.

- **Screenshot of micro xrce client**   
[image]

- **Screenshots of xrce agent with the flag :** `-v6`
[image]
[image]

- **Screenshot of ros2 with** `ros2 topic echo /NumTopic`
[image]
(no response on the ros2 end)

pablogs9, 2021-07-05:
Sorry I did not write well the type name mangling try to use: `std_msgs::msg::dds_::Int32_`

closed 2021-07-05 by pablogs9

arshPratap, 2021-07-15:
@pablogs9 for the help!! Its properly working now