eProsima/Micro-XRCE-DDS-Client #253: session.c: 'time' shadows a global declaration

gavanderhoorn, 2021-07-05:
This is related to #252 (in that it's the same compiler/environment).

The `time` argument to `uxr_sync_session(..)` causes a compiler warning in my dev env about it shadowing a "global declaration". That global declaration appears to be [time_t time(time_t *t);](https://linux.die.net/man/2/time):

https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/69134a9ba97f98f3cfad1588af816a6a22133116/src/c/core/session/session.c#L440-L442

I'm not sure whether this warning would happen on other platforms as well -- it might be my compiler is too old to properly isolate things fi -- but the easiest fix to avoid the warning would be to rename the argument (the header provided by the SDK cannot be changed).

Would it be acceptable to rename the `time` argument to something else?

That argument is passed on to `run_session_until_sync(..)` which actually calls it `timeout`. It's not used anywhere else in `uxr_sync_session(..)`.

gavanderhoorn, 2021-07-05:
Perhaps `timeout` could be a good name. The doxygen comment for the argument seems to already describe it as such:

https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/69134a9ba97f98f3cfad1588af816a6a22133116/include/uxr/client/core/session/session.h#L438-L443

pablogs9, 2021-07-05:
Nice, I'm ok with the change to `timeout`. Can you open a PR and close this issue?

gavanderhoorn, 2021-07-05:
Against `develop` again?

pablogs9, 2021-07-05:
Yes please

pablogs9, 2021-07-05:
Closing since patch merged

commit 148a3db1 (2021-07-05): session: don't shadow global time(..). Fix #253 (#254)
closed 2021-07-05 by pablogs9