eProsima/Micro-XRCE-DDS-Client #268: Field has incomplete type with UDP transport

coalman321, 2021-10-03:
Hi,
I am trying to compile this library as part of micro-ros, but have run into an issue during the build process. The target board is armhf with some custom compiler configurations that I am not entirely aware of. I do also know it is older than the mainline version of GCC by at least 2 years. 

Currently the build fails with the error
`In file included from /home/coalman321/code/micro_ros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/include/uxr/client/transport.h:30:0,
                 from /home/coalman321/code/micro_ros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/include/uxr/client/util/ping.h:28,
                 from /home/coalman321/code/micro_ros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/src/c/util/ping.c:1:
/home/coalman321/code/micro_ros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/include/uxr/client/profile/transport/ip/udp/udp_transport.h:37:27: error: field 'platform' has incomplete type
     struct uxrUDPPlatform platform;
                           ^~~~~~~~
make[2]: *** [CMakeFiles/microxrcedds_client.dir/build.make:261: CMakeFiles/microxrcedds_client.dir/src/c/util/ping.c.obj] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/coalman321/code/micro_ros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/src/c/profile/transport/ip/udp/udp_transport_internal.h:23:0,
                 from /home/coalman321/code/micro_ros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/src/c/profile/transport/ip/udp/udp_transport.c:2:
/home/coalman321/code/micro_ros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/include/uxr/client/profile/transport/ip/udp/udp_transport.h:37:27: error: field 'platform' has incomplete type
     struct uxrUDPPlatform platform;
                           ^~~~~~~~
/home/coalman321/code/micro_ros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/src/c/profile/transport/ip/udp/udp_transport.c: In function 'uxr_close_udp_transport':`

Any help in fixing this would be greatly appreciated.

pablogs9, 2021-10-04:
You are not configuring the library well, could you share your `colcon.meta` or the CMake args that you are using? Also, detail the transport that you want to use to communicate client with the agent.

coalman321, 2021-10-04:
colcon.meta : https://gist.github.com/coalman321/302fa34fa7ca37d07bfce5f21e27490a
toolchain.cmake : https://gist.github.com/coalman321/c568b9d817791a531f2eb09c110a415f
The desired transport is UDP and or TCP to the agent. The system is running a soft realtime linux kernel from National Instruments

pablogs9, 2021-10-04:
Disable UDP profile in the Micro XRCE-DDS Client configuration if you don't want POSIX UDP support: https://gist.github.com/coalman321/302fa34fa7ca37d07bfce5f21e27490a#file-colcon-meta-L33

coalman321, 2021-10-04:
If I set that to off, it does compile, but then that would leave the udp network stack configuration up to me correct? I would like to stick with the posix version if possible. Am I missing some kind of tool chain flag to enable this? Or is an armv7 Linux kernel running the real-time scheduler not posix compliant?

coalman321, 2021-10-04:
As a side note, I tested the same colcon.meta file with the custom transport and stream framing off. Changing e

pablogs9, 2021-10-05:
On which platform are you using micro-ROS?

coalman321, 2021-10-05:
The platform is not an officially supported one from the list. I am attempting to build a static library for a Xilinx armv7 board with an integrated FPGA. The device is called the Roborio and it is produced by National Instruments (NI). It is running a 4.x Linux kernel with the real-time scheduler patch enabled. (I’ll add the output of uname -a when I can access the board later today) It is unable to run a fully fledged version of ros2 due to only having 256 Mb of ram, but it does make a very powerful for micro ros to run on.

coalman321, 2021-10-05:
Output of uname -a:
Linux roboRIO-4145-FRC 4.14.87-rt49-cg-7.0.0f0-xilinx-zynq-189 #1 SMP PREEMPT RT Fri Mar 15 19:11:24 UTC 2019 armv7l GNU/Linux

pablogs9, 2021-10-05:
In this case, reenable the UDP flag (`UCLIENT_PROFILE_UDP`) and make sure that the CMake in the XRCE-DDS Client is setting [this flag](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/69134a9ba97f98f3cfad1588af816a6a22133116/CMakeLists.txt#L129) and [this](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/69134a9ba97f98f3cfad1588af816a6a22133116/include/uxr/client/transport.h#L26) is being included.

coalman321, 2021-10-05:
How should I check those second two? I am not overly familliar with cmake and the uRos build system and hot to get set variables and flags out. Though, I think I can manually force them with the toolchain file, but I am not sure if that would be desirable in this instance.

pablogs9, 2021-10-05:
Yes, setting them in the toolchain should be ok.