eProsima/Micro-XRCE-DDS-Client #281: Client can't connect to the agent

amfern, 2021-11-13:
Hi,

I am using [arduino_micro_ros](https://github.com/micro-ROS/micro_ros_arduino) for a while now without updating, and it works great.
Today, I compiled the new version of arduino_micro_ros, but the client wouldn't connect to the micro ros agent anymore, i run the agent as docker `microros/micro-ros-agent:galactic`.

I traced the issue to commit 264191a767380180b7ab023f6be91c8fd2abbc1b in this repo. If I compile the code before that commit, the connection works without issues.
I also downgraded https://github.com/micro-ROS/rmw_microxrcedds  to commit b996df14b72aa67bf40b2e159414fa86e72c5f14 so it would compile.

amfern, 2021-11-13:
Doing further git-bisect this lead me to commits cb4f5220b5666be270ac29820ccd3e7a19f267f1 + e40f4d6950598d4b192665c029484ea45727e05a in https://github.com/micro-ROS/rmw_microxrcedds.
So I was wrong in the initial post. The issue is not in the  Micro-XRCE-DDS-Client

pablogs9, 2021-11-15:
Please open this issue in a micro-ROS repo and detail your environment (board, transport, configuration, sample code...).

closed 2021-11-15 by pablogs9