eProsima/Micro-XRCE-DDS-Client #286: Byte alignment on message larger than UCLIENT_UDP_TRANSPORT_MTU

ted-miller, 2021-11-16:
I have encountered an issue using MicroROS with a "large" message. I am using the v2.0.0 release of Micro XRCE DDS Client.

The symptom is very similar to [this issue](https://github.com/eProsima/Micro-XRCE-DDS-Agent/issues/99). 

I am publishing a TFMessage message that is larger than the default `UCLIENT_UDP_TRANSPORT_MTU` of 512 bytes.
The final transform in the sequence has garbage data in it.
```
    frame_id: flange_r2
  child_frame_id: tcp_1
  transform:
    translation:
      x: -1.3398716191361611e+248
      y: -48.70349877276849
      z: 2.4927293553960274e-240
    rotation:
      x: -1.4304829656843133e+239
      y: 1.1594275270700436e+176
      z: -1.5472557014055978e+293
      w: 1.75056776064e-313
```

<details><summary>Full output from 'ros2 echo topic /tf'</summary>
<p>

```
transforms:
header:
    stamp:
      sec: 1637074677
      nanosec: 490800000
    frame_id: world
  child_frame_id: base_r1
  transform:
    translation:
      x: 0.0
      y: 0.0
      z: 0.0
    rotation:
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
header:
    stamp:
      sec: 1637074677
      nanosec: 490800000
    frame_id: base_r1
  child_frame_id: flange_r1
  transform:
    translation:
      x: 0.8949999999999999
      y: 0.0
      z: 0.814
    rotation:
      x: 0.7071068613706294
      y: -1.6036819105398803e-07
      z: 0.7071067010024201
      w: 1.6036822742467945e-07
header:
    stamp:
      sec: 1637074677
      nanosec: 490800000
    frame_id: flange_r1
  child_frame_id: tcp_0
  transform:
    translation:
      x: 0.023219
      y: -0.089918
      z: 0.394207
    rotation:
      x: -0.006538292096494784
      y: 0.17387234808898264
      z: -0.9842079497518248
      w: 0.03256330683137005
header:
    stamp:
      sec: 1637074677
      nanosec: 493000000
    frame_id: world
  child_frame_id: base_r2
  transform:
    translation:
      x: 1.4285139999999998
      y: 0.097566
      z: 0.053146
    rotation:
      x: 0.02331814105334959
      y: -0.005115648592001999
      z: -0.9983376600983214
      w: 0.05245961176665228
header:
    stamp:
      sec: 1637074677
      nanosec: 493000000
    frame_id: base_r2
  child_frame_id: flange_r2
  transform:
    translation:
      x: 0.8949999999999999
      y: 0.0
      z: 0.814
    rotation:
      x: 0.7071068613706294
      y: -1.6036819105398803e-07
      z: 0.7071067010024201
      w: 1.6036822742467945e-07
header:
    stamp:
      sec: 1637074677
      nanosec: 493000000
    frame_id: flange_r2
  child_frame_id: tcp_1
  transform:
    translation:
      x: -1.3398716191361611e+248
      y: -48.70349877276849
      z: 2.4927293553960274e-240
    rotation:
      x: -1.4304829656843133e+239
      y: 1.1594275270700436e+176
      z: -1.5472557014055978e+293
      w: 1.75056776064e-313

```

</p>
</details>

I reviewed a wireshark capture to verify whether the data was corrupted on the network. When I compare the contents of the wireshark data, I can see that the final transform hit the edge of the packet size and was placed into a second packet. But, if I look at the data ([with this calculator](https://gregstoll.com/~gregstoll/floattohex/)), I can see that the data in the packet is correct.

It appears that the deserializer is offset by 4 bytes when reading the sequence of doubles. 

[wireshark.zip](https://github.com/eProsima/Micro-XRCE-DDS-Client/files/7549285/wireshark.zip)

<details><summary>Wireshark summary</summary>
<p>

```
No.     Time           Source                Destination           Length Info                                                            Protocol Text
    167 5.821712       192.168.1.31          192.168.1.178         550    55880→2018 Len=508                                            UDP      \357\277\275\357\277\2757

Frame 167: 550 bytes on wire (4400 bits), 550 bytes captured (4400 bits) on interface 0
Ethernet II, Src: Ricoh_c7:12:4e (00:26:73:c7:12:4e), Dst: WistronI_12:28:60 (54:ee:75:12:28:60)
Internet Protocol Version 4, Src: 192.168.1.31, Dst: 192.168.1.178
User Datagram Protocol, Src Port: 55880, Dst Port: 2018
Data (508 bytes)

0000  81 80 37 00 0d 01 f4 01 07 01 3c 02 00 36 00 35   ..7.......<..6.5
0010  06 00 00 00 e6 c6 93 61 80 35 91 31 06 00 00 00   .......a.5.1....
0020  77 6f 72 6c 64 00 69 6e 08 00 00 00 62 61 73 65   world.in....base
0030  5f 72 31 00 74 5f 31 00 00 00 00 00 00 00 00 00   _r1.t_1.........
0040  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
0050  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
0060  00 00 00 00 00 00 00 00 00 00 00 00 00 00 f0 3f   ...............?
0070  e6 c6 93 61 80 35 91 31 08 00 00 00 62 61 73 65   ...a.5.1....base
0080  5f 72 31 00 0a 00 00 00 66 6c 61 6e 67 65 5f 72   _r1.....flange_r
0090  31 00 00 00 00 00 00 00 a3 70 3d 0a d7 a3 ec 3f   1........p=....?
00a0  00 00 00 00 00 00 00 00 3f 35 5e ba 49 0c ea 3f   ........?5^.I..?
00b0  5f a6 8b 91 9e a0 e6 3f 66 85 bc 86 35 86 85 be   _......?f...5...
00c0  a0 cf 72 3b 9e a0 e6 3f 17 cc a2 d8 35 86 85 3e   ..r;...?....5..>
00d0  e6 c6 93 61 80 35 91 31 0a 00 00 00 66 6c 61 6e   ...a.5.1....flan
00e0  67 65 5f 72 31 00 00 00 06 00 00 00 74 63 70 5f   ge_r1.......tcp_
00f0  30 00 00 00 00 00 00 00 e3 52 95 b6 b8 c6 97 3f   0........R.....?
0100  c4 5c 52 b5 dd 04 b7 bf 64 ac 36 ff af 3a d9 3f   .\R.....d.6..:.?
0110  5e 45 9e 6b e5 c7 7a bf bb 46 5c f8 72 41 c6 3f   ^E.k..z..F\.rA.?
0120  ac b6 94 ab a1 7e ef bf a9 db c7 43 23 ac a0 3f   .....~.....C#..?
0130  e6 c6 93 61 40 c7 b2 31 06 00 00 00 77 6f 72 6c   ...a@..1....worl
0140  64 00 00 00 08 00 00 00 62 61 73 65 5f 72 32 00   d.......base_r2.
0150  e0 0c fe 7e 31 db f6 3f dd 97 33 db 15 fa b8 3f   ...~1..?..3....?
0160  0c 91 d3 d7 f3 35 ab 3f 78 1d e9 f4 b5 e0 97 3f   .....5.?x......?
0170  9b 5a 68 76 25 f4 74 bf 3a 99 0f d2 61 f2 ef bf   .Zhv%.t.:...a...
0180  03 19 cc 79 fc db aa 3f e6 c6 93 61 40 c7 b2 31   ...y...?...a@..1
0190  08 00 00 00 62 61 73 65 5f 72 32 00 0a 00 00 00   ....base_r2.....
01a0  66 6c 61 6e 67 65 5f 72 32 00 0b 55 fa c1 da 40   flange_r2..U...@
01b0  a3 70 3d 0a d7 a3 ec 3f 00 00 00 00 00 00 00 00   .p=....?........
01c0  3f 35 5e ba 49 0c ea 3f 5f a6 8b 91 9e a0 e6 3f   ?5^.I..?_......?
01d0  66 85 bc 86 35 86 85 be a0 cf 72 3b 9e a0 e6 3f   f...5.....r;...?
01e0  17 cc a2 d8 35 86 85 3e e6 c6 93 61 40 c7 b2 31   ....5..>...a@..1
01f0  0a 00 00 00 66 6c 61 6e 67 65 5f 72               ....flange_r

No.     Time           Source                Destination           Length Info                                                            Protocol Text
    168 5.821713       192.168.1.31          192.168.1.178         126    55880→2018 Len=84                                             UDP      \357\277\275\357\277\2758

Frame 168: 126 bytes on wire (1008 bits), 126 bytes captured (1008 bits) on interface 0
Ethernet II, Src: Ricoh_c7:12:4e (00:26:73:c7:12:4e), Dst: WistronI_12:28:60 (54:ee:75:12:28:60)
Internet Protocol Version 4, Src: 192.168.1.31, Dst: 192.168.1.178
User Datagram Protocol, Src Port: 55880, Dst Port: 2018
Data (84 bytes)

0000  81 80 38 00 0d 03 4c 00 32 00 00 15 06 00 00 00   ..8...L.2.......
0010  74 63 70 5f 31 00 00 00 aa d0 40 2c 9b 39 b8 bf   tcp_1.....@,.9..
0020  c3 29 73 f3 8d e8 6e 3f 0c 5a 48 c0 e8 f2 d7 3f   .)s...n?.ZH....?
0030  1e 9f 30 0e 74 1d c6 3f c0 f7 95 f1 d3 37 bd bf   ..0.t..?.....7..
0040  6d 4c 7d 64 4c 7c e4 bf 75 02 cf fc 6c ac e7 3f   mL}dL|..u...l..?
0050  08 00 00 00                                       ....

```

</p>
</details>

Interestingly, the `child_frame_id` was also contained in the second packet, but it was parsed correctly. The problem doesn't occur until it tries to read the sequence of doubles. 

I am able to work around this issue for now by simply increasing `UCLIENT_UDP_TRANSPORT_MTU` to keep everything contained in a single packet.

pablogs9, 2021-11-17:
Could you provide code for replicating this issue?

gavanderhoorn, 2021-11-17:
What sort of code are you looking for @pablogs9 (as in: how extensive)? The specific parts for the message initialisation and publication, or a complete Micro-ROS application?

Essentially any message which -- when serialised -- is larger than the configured MTU seems to exhibit this behaviour.

pablogs9, 2021-11-17:
> What sort of code are you looking for @pablogs9 (as in: how extensive)? The specific parts for the message initialisation and publication, or a complete Micro-ROS application?

A code where this TF sequence message is being created using the micro-ROS API and being published in order to:
1. Check if everything is being configured correctly
2. Replicate the same environment and check what is doing wrong with the message type support or the XRCE-DDS fragmentation.

> Essentially any message which -- when serialised -- is larger than the configured MTU seems to exhibit this behaviour.

We have a [complete test suite](https://github.com/eProsima/Micro-XRCE-DDS) that runs nightly and as CI and we have not found any error when using XRCE-DDS fragmentation (aka any message larger than MTU). So, we need the erroneous specific use case where the library is failing.

> Edit: and to make sure: the UDP transport is compatible with / takes care of framing when `UCLIENT_PROFILE_STREAM_FRAMING` is `ON`, correct? Or is more required?

UDP transport is a packet-based transport and has nothing to do with XRCE-DDS Stream framing.

pablogs9, 2021-11-17:
Found a weird behaviour in micro-CDR and micro-ROS type support when handling types of size 8 B at the end of a message.

I will inform and provide a patch ASAP.

gavanderhoorn, 2021-11-17:
Is eProsima/Micro-CDR#69 the patch WIP @pablogs9 ?

pablogs9, 2021-11-17:
Yes, working on it, but not solved yet.

ted-miller, 2021-11-17:
> Could you provide code for replicating this issue?

I will try to have something for you tomorrow afternoon. (US time)

ted-miller, 2021-11-18:
Here is some simplified code to reproduce the issue.
https://github.com/ted-miller/issue286/blob/84d814263058f973a7699fe1f671fa5771f37c02/main.c

gavanderhoorn, 2021-11-22:
@pablogs9: based on the work in https://github.com/eProsima/Micro-CDR/pull/69 it looks like there was indeed something external to the / our application which was the cause of the problem reported here, correct?

pablogs9, 2021-11-22:
That's right. We are having problems with the CDR buffer alignment when fragmentation happens. eProsima/Micro-CDR#69 seems to solve the problem but we are still testing.

Could you test this branch and check if it works in your use case?

gavanderhoorn, 2021-11-23:
We'll try to do that @pablogs9.

ted-miller, 2021-11-23:
I have just verified that https://github.com/eProsima/Micro-CDR/pull/69 works with our application. 

Thank you for this fix.

pablogs9, 2021-11-23:
I'll close this when eProsima/Micro-CDR#69 is merged. Thanks for the tests!

closed 2021-11-25 by pablogs9