eProsima/Micro-XRCE-DDS-Client #306: SHM feature On NuttX

PierrePeng, 2021-12-18:
Hi,

I am confused about the share memory feature On Nuttx.

1.  Can I publish a topic from program A and receive it both from program B and program C?
2.  Should I create sessions/streams/entities/publishers/subscribers for each topic that I want to publish/subscribe to in a single program?

Thanks!

pablogs9, 2021-12-20:
> * Can I publish a topic from program A and receive it both from program B and program C?

Yes if those programs share the same memory and address space. SHM in XRCE is implemented as a naive approach that does not take into account process memory protection.

> * Should I create sessions/streams/entities/publishers/subscribers for each topic that I want to publish/subscribe to in a single program?

One session can have multiple streams, and those streams and the session can handle multiple pubs, subs, and topics.

PierrePeng, 2021-12-22:
@pablogs9 Thanks for your reply!

> One session can have multiple streams, and those streams and the session can handle multiple pubs, subs, and topics.

Are there any tutorials or examples about that?

pablogs9, 2021-12-22:
You have all the available examples here: https://github.com/eProsima/Micro-XRCE-DDS-Client/tree/master/examples

closed 2025-09-25 by 4ntn