eProsima/Micro-XRCE-DDS-Client #324: Not checking return value of setsockopt leads to silent failure when error occurs

k7g03z, 2022-08-03:
In the following line, it is attempted to set the receive timeout by calling the `setsockopt` with the appropriate parameters

https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/6f4edd25c0984fd30bf3bf84e2d1d33af03f24a1/src/c/profile/discovery/transport/udp_transport_datagram_posix_nopoll.c#L70

When setting up Micro-ROS on a new hardware platform, the executor did not function properly. After lots of debugging, we found out that the platform provider did not set the define in their generated `lwipopts.h`:

`#define LWIP_SO_RCVTIMEO 1 ` 

When the define is not set, the LWIP stack does not provide the receive timeout functionality, and `setsockopt` returns `-1`. 

The following call:
https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/6f4edd25c0984fd30bf3bf84e2d1d33af03f24a1/src/c/profile/discovery/transport/udp_transport_datagram_posix_nopoll.c#L72

then waits indefinitely, blocking any further Micro-ROS executor functionality.

Since the above code does not check the error, the user has no idea why the executor runs once and then nothing else happens..

pablogs9, 2022-08-04:
Could you take a look at this PR and provide some feedback: https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/325 ??

Thanks a lot for the report, very useful.

Acuadros95, 2022-08-16:
Closing as https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/325 merged.

Please reopen if needed, thanks for the catch!

closed 2022-08-16 by Acuadros95