eProsima/Micro-XRCE-DDS-Client #326: Participant msg failed

VasuDhevan22, 2022-08-30:
Hi,
Currently am porting xrcedds client library to the mpc5744 power pc architecture. Code compiling was fine. We can able to make session init while sending participant msg am getting error from agent side. like de-serialization error.
Just analyzed the frame compare to the linux client packet there was a some issue in framing.
We are configured - BIG ENDIAN format

**Participant frame's:**

Jetson nano client frame:
0000: 81 80 00 00 01 05 68 00 00 0A 00 11 01 02 CD **7E 59 00 00 00** 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 64 65 66 61 75 6C 74 5F 78 72 63 65 5F 70
0040: 61 72 74 69 63 69 70 61 6E 74 3C 2F 6E 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63
0060: 69 70 61 6E 74 3E 3C 2F 64 64 73 3E 00 **CA** 00 00

MPC5744 client frame:
0000: 81 80 00 00 01 04 68 00 00 0A 00 11 01 02 0**0 00 00 00 00 59** 3C 64 64 73 3E 3C 70 61 72 74 69 63
0020: 69 70 61 6E 74 3E 3C 72 74 70 73 3E 3C 6E 61 6D 65 3E 64 65 66 61 75 6C 74 5F 78 72 63 65 5F 70
0040: 61 72 74 69 63 69 70 61 6E 74 3C 2F 6E 61 6D 65 3E 3C 2F 72 74 70 73 3E 3C 2F 70 61 72 74 69 63
0060: 69 70 61 6E 74 3E 3C 2F 64 64 73 3E 00 **00** 00 00

There is a frame mismatch in mpc5744.
Please help me to resolve this issue.

pablogs9, 2022-09-05:
Could you provide code for replicating this issue?

king318hua, 2022-12-29:
Hi,
Currently am porting xrcedds client library to the mpc5748G power pc architecture. Code compiling was fine. We can able to make session init while sending participant msg am getting error from agent side. like de-serialization error. but  i send 128 packages, i will stoped.

king318hua, 2022-12-29:
> Hi, Currently am porting xrcedds client library to the mpc5744 power pc architecture. Code compiling was fine. We can able to make session init while sending participant msg am getting error from [...]

i  force UCDR_BIG_ENDIANNESS，it will ok, but Another problem appears.

pablogs9, 2022-12-30:
Which other problem appears?

4ntn, 2025-09-25:
Closing due to inactivity

closed 2025-09-25 by 4ntn