eProsima/Micro-XRCE-DDS-Client #329: How to use the "uxr_buffer_create_requester_bin" function

GarMingLi, 2022-11-04:
A client uses the "uxr_buffer_create_replier_xml" function to create a replier. Another client uses the "uxr_buffer_create_requester_bin" function to create a requester.
The "uxr_buffer_create_requester_bin" function contains two parameters: "request_topic_name" and "reply_topic_name". If they are set to "", the entity cannot be created successfully. How to use this function to create a requester to communicate with the replier?

pablogs9, 2022-11-07:
Well, as stated in [the documentation](https://micro-xrce-dds.docs.eprosima.com/en/latest/client_api.html#entities-bin-api), those two arguments represent the name of the pair of topics used to generate the service. They cannot be empty.

closed 2022-11-07 by pablogs9

GarMingLi, 2022-11-07:
> 
But in the example, these two parameters are empty。

pablogs9, 2022-11-07:
Hello @GarMingLi could you point out where is this example? This should be fixed.

GarMingLi, 2022-11-07:
> 

Hello @pablogs9 On“[On“ https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/master/examples/SharedMemoryReqRep/main.c ”In this link](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/master/examples/SharedMemoryReqRep/main.c)”In this link.
```
    // Create requester
    uxrObjectId requester_id = uxr_object_id(0x01, UXR_REQUESTER_ID);
    uxrQoS_t qos = {
        .reliability = UXR_RELIABILITY_RELIABLE, .durability = UXR_DURABILITY_TRANSIENT_LOCAL,
        .history = UXR_HISTORY_KEEP_LAST, .depth = 1
    };

    uxr_buffer_create_requester_bin(&session, output_besteffort, requester_id, participant_id, "shared_memory_reqres",
            "req_type", "res_type", "", "", qos, UXR_REPLACE);
    uxr_set_request_callback(&session, on_request, NULL);

    // Create replier
    replier_id = uxr_object_id(0x01, UXR_REPLIER_ID);
    uxr_buffer_create_replier_bin(&session, reliable_in, replier_id, participant_id, "shared_memory_reqres", "req_type",
            "res_type", "", "", qos, UXR_REPLACE);
```

pablogs9, 2022-11-08:
You are using the shared memory feature, in this case, no topics are being created, so those strings can be empty. Have you built the library with `UCLIENT_PROFILE_SHARED_MEMORY` set to `ON`?

GarMingLi, 2022-11-08:
> 

I am not using the "UCLIENT_PROFILE_SHARED_MEMORY" feature. I understand now, thanks! @pablogs9