eProsima/Micro-XRCE-DDS-Client #333: Miss the 1st one

jinfeihan57, 2022-11-10:
I am a new user of dds. when I run example PublishHelloWorld->SubscribeHelloWorld or CustomTransports->SubscribeHelloWorld. 
The first topic is lost almost every time. I want to know WHY? and how to fix this. Thanks.

pablogs9, 2022-11-10:
You can create your entities with transient local QoS and a certain amount of history in order to ensure the reception of all samples. You can learn more about QoS in the Fast DDS documentation: https://fast-dds.docs.eprosima.com/en/latest/fastdds/rtps_layer/rtps_layer.html?highlight=TRANSIENT_LOCAL#setting-the-data-durability-kind

jinfeihan57, 2022-11-11:
> You can create your entities with transient local QoS and a certain amount of history in order to ensure the reception of all samples. You can learn more about QoS in the Fast DDS documentation: [...]

VOLATILE (default): Messages are discarded as they are sent. If a new Reader matches after message n, it will start received from message n+1.
I start the Subscribe first, when MicroXRCEAgent found Subscribe, then I start Publish. this should not topic miss.

pablogs9, 2022-11-11:
Ok, provide feedback and close if the issue is solved.

jinfeihan57, 2022-11-11:
It did not solve my issue.

pablogs9, 2022-11-11:
What else do you need beyond modifying the QoS of your entities?

jinfeihan57, 2022-11-11:
[image]

pablogs9, 2022-11-11:
Well, when you change the code to apply the required QoS, then, provide feedback and close if the issue is solved.

closed 2025-09-25 by 4ntn