eProsima/Micro-XRCE-DDS-Client #334: Getting dlfcn.h file not found error when Shared Memory Enabled and also platform error when Multi Thread Enabled

ManvithRaj, 2022-12-05:
Hi  @pablogs9 ,

Could you please look at below issue and let me know what might be the reason ?

I am using Microros workspace with  ESP32 Firmware created on my Linux Ubuntu virutal machine .I have enabled Shared Memory in colcon.meta of XRCE-DDS Client file as suggested.

"-DUCLIENT_PROFILE_SHARED_MEMORY=ON"
 
But if i try to build the code i am getting Error as fatal error: dlfcn.h: No such file or directory

Could you help in this ?

thank you in advance.

pablogs9, 2022-12-05:
Hello this is not a micro-ROS repo, please use a micro-ROS repo.

`dlfcn.h` does not seems to be a micro-ROS header. Please provide instructions for replicating.

closed 2022-12-05 by pablogs9

ManvithRaj, 2022-12-05:
Hi @pablogs9 

Since the issue related to enabling shared memory in Micro-XRCE-DDS-Client so i have posted here.

I have followed micro-ros installation and build procedure from below link which has all the explanation

https://medium.com/@SameerT009/connect-esp32-to-ros2-foxy-5f06e0cc64df

I have changed colcon.meta file from path  microros_ws/firmware/mcu_ws with shared memory enabled as below  image

[image]

getting below error in shared_library.c as below 
[image]

please help me in this issue.

pablogs9, 2022-12-05:
This section shall not be build: https://github.com/micro-ROS/rcutils/blob/humble/src/shared_library.c#L24-L55
Since `RCUTILS_NO_FILESYSTEM` is defined.

ManvithRaj, 2022-12-05:
Hi @pablogs9 

Thank you for the reply.

I tried defining  #define RCUTILS_NO_FILESYSTEM in same file to bypass this but now ending up with below platform issue

[image]

could you please help in this ?

pablogs9, 2022-12-05:
If you are using ESP32 you shall use these instructions to get it working: https://github.com/micro-ROS/micro_ros_espidf_component#example

When it works with those instructions you shall try to enable shared memory operation. Those errors seem to be not related to shared memory.

ManvithRaj, 2022-12-05:
hi @pablogs9 .

I am parallelly trying also the method you said which is enabling share memory operation with Micro-ros Middleware selecting as XRCE-DDS in menuconfig with example you said link (https://github.com/micro-ROS/micro_ros_espidf_component#example)

But here i am ending the linking issue which i have already raised issue here https://github.com/micro-ROS/micro_ros_espidf_component/issues/167

Could you please check this and help in this ?