eProsima/Micro-XRCE-DDS-Client #349: Enabling BUILD_DOCUMENTATION results in Doxygen Error

Ryanf55, 2023-03-04:
I am having an internal issue in the library. I would like to use the `CALL_GRAPH` to inspect what my calls are doing and where to look. 

When I enable `BUILD_DOCUMENATION` with CMake, it fails to build docs. 

Branch: `develop` hash `85d83a925072f4fb377a41828af741bf3de54b67`
OS: Linux Ubuntu 22.04 
Command: `cmake -S . -B build -DBUILD_DOCUMENTATION=ON && cmake --build build`
```
Doxygen version used: 1.9.1
Notice: Output directory '/home/ryan/Documents/ardu_ws/src/ardupilot/modules/Micro-XRCE-DDS-Client/build/doxygen' does not exist. I have created it for you.
Parsing layout file /home/ryan/Documents/ardu_ws/src/ardupilot/modules/Micro-XRCE-DDS-Client/utils/doxygen/pages/DoxygenLayout.xml...
/home/ryan/Documents/ardu_ws/src/ardupilot/modules/Micro-XRCE-DDS-Client/utils/doxygen/pages/DoxygenLayout.xml:21: error: the type 'dirs' is not supported for the entry tag within a navindex! Check your layout file!
 (warning treated as error, aborting now)
Exiting...
gmake[5]: *** [CMakeFiles/doxygen.dir/build.make:71: CMakeFiles/doxygen] Error 1
gmake[4]: *** [CMakeFiles/Makefile2:139: CMakeFiles/doxygen.dir/all] Error 2
gmake[3]: *** [Makefile:136: all] Error 2
gmake[2]: *** [CMakeFiles/uclient.dir/build.make:86: uclient-prefix/src/uclient-stamp/uclient-build] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:111: CMakeFiles/uclient.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2
```

Acuadros95, 2023-03-06:
Hi @Ryanf55, this PR should solve the issue: #352

Ryanf55, 2023-03-07:
Ok. I tried on the PR, there is undocumented errors. 

```
/home/ryan/Development/ardu_ws/src/ardupilot/modules/Micro-XRCE-DDS-Client/include/uxr/client/core/session/session.h:140: error: The following parameter of uxrOnBuffersFull(struct uxrSession *session, void *args) is not documented:
  parameter 'args' (warning treated as error, aborting now)
Exiting...
gmake[2]: *** [CMakeFiles/doxygen.dir/build.make:71: CMakeFiles/doxygen] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:139: CMakeFiles/doxygen.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2
ryan@ryan-B650:~/Development/ardu_ws/src/ardupilot/modules/Micro-XRCE-DDS-Client$ git log
commit fe83fd082e82ab1aeed75829e04c10207df74576 (HEAD -> develop, origin/ros2, origin/foxy, origin/develop)
Author: Antonio Cuadros <49162117+Acuadros95@users.noreply.github.com>
Date:   Mon Mar 6 12:22:55 2023 +0100

    Fix doxygen warning (#352)
    
    * Remove outdated navindex tag
    
    Signed-off-by: acuadros95 <acuadros1995@gmail.com>
    
    * Revert unwanted removal
    ```
    
    

Since warnings are treated as errors, I still can't run.

Acuadros95, 2023-03-07:
Had to update my doxy version to get the warning.
Fixed here: https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/353

Ryanf55, 2023-03-07:
Thanks that worked. 
[image]

closed 2023-03-07 by Ryanf55