eProsima/Micro-XRCE-DDS-Client #374: Multiple Clients cannot connect to same CAN Agent

PeterRose710, 2023-11-14:
The use of CAN bus as transport suggests that there can be more than two participants on the network, this ought to be possible per XRCE-DDS spec, however trying to add a second client results in failure where the initial client gets deleted when a session was already initialized.  
The result is that name of second client is shown at ros2 host but none of the topics list up.  
[image]

pablogs9, 2023-11-14:
Please share the Micro XRCE-DDS Agent as text and using the `-v6` flag. Indicate when are you connecting the first and the second CAN XRCE Client

PeterRose710, 2023-11-14:
Here we go...
[CanAgent-Client-Drop.txt](https://github.com/eProsima/Micro-XRCE-DDS-Client/files/13350218/CanAgent-Client-Drop.txt)
Thanks

PeterRose710, 2023-11-20:
it seems it's not an agent issue. I got it working now, the issue was that on my test code can frame ID was always the same number, thus causing a conflict since the ID sets the endpoint at session manager CanEndPoint(can_id) which sets a map at https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/40954c2379dfcb2bfa9f6ba22146c57c6742b5b7/include/uxr/agent/transport/SessionManager.hpp#L62
So using individual can ID separated the sessions and now agents handles both clients simultaneously.

pablogs9, 2023-11-20:
Nice to hear that, should I close the issue as solved?

PeterRose710, 2023-11-20:
I think so, there is still an issue with sequencing of the initialization, if both clients start simultaneously, only one starts correctly, the other has to reset to get initialized. But this might be not related to this.

pablogs9, 2023-11-20:
Ok, please open a new issue if you find that this init error is related to XRCE.

Thanks!

closed 2023-11-20 by pablogs9