eProsima/Micro-XRCE-DDS-Client #377: Topic published from MCU client cannot be echoed by ROS2

Pansamic, 2023-12-20:
# Problem

`ros2 topic echo /imu/data` **shows nothing**, but `ros2 topic list` shows correct topic `/imu/data`.

# Detail

* message source files are generated by Micro-XRCE-DDS-Gen. dependent IDL files are collected from `/opt/ros/humble/share/xxxx_msgs/` folders.
* use `rt/imu/data` for dds topic name and `sensor_msgs::dds_::Imu_` for dds data type.
* topic, reply, request, status callbacks are not registered, since I just publish a message.
* set client domain id to 3, and micro-ros-agent env `export ROS_DOMAIN_ID=3` so Domain ID matches.
* `ros2 node list` shows client participant name **as expected**.
* `ros2 topic list` shows client topic name **as expected**.
* use micro-ros agent, `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -v5`
    ```
    [1703074634.781723] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM1, error 2, waiting for connection...
    [1703074635.402391] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
    [1703074641.064788] info     | Root.cpp           | create_client            | create                 | client_key: 0x465119D3, session_id: 0x81
    [1703074641.064893] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x465119D3, address: 0
    [1703074641.082615] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x465119D3, participant_id: 0x001(1)
    [1703074641.082697] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x465119D3, topic_id: 0x000(2), participant_id: 0x001(1)
    [1703074641.082729] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x465119D3, publisher_id: 0x000(3), participant_id: 0x001(1)
    [1703074641.083123] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x465119D3, datawriter_id: 0x000(5), publisher_id: 0x000(3)
    [1703074553.476758] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x465119D3, len: 13
    [1703074553.481617] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x465119D3, len: 332
    [1703074553.481697] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 320
    [1703074553.481722] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x465119D3, len: 13
    [1703074553.486604] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x465119D3, len: 332
    [1703074553.486712] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 320
    [1703074553.486738] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x465119D3, len: 13
    [1703074553.491617] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x465119D3, len: 332
    [1703074553.491708] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 320
    [1703074553.491787] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x465119D3, len: 13
    [1703074553.496640] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x465119D3, len: 332
    [1703074553.496744] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 320
    [1703074553.496828] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x465119D3, len: 13
    [1703074553.501673] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x465119D3, len: 332
    [1703074553.501767] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 320
    [1703074553.501842] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x465119D3, len: 13
    [1703074553.506592] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x465119D3, len: 332
    [1703074553.506684] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 320
    [1703074553.506710] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x465119D3, len: 13
    ```
main code looks like:
```c
int main(void)
{
    uxrce_client_init();
    create_publisher_imu();

    while(1)
    {
        // `imu_update()` and `load_imu_data_to_msg()` are simplified fake function of real ones.
        imu_update();
        load_imu_data_to_msg(&msg_imu);
        
        msg_publish(&msg_imu);
}
```
Details of `uxrce_client_init()`, etc.
```c
uxrQoS_t reliable_qos = {
    .durability = UXR_DURABILITY_VOLATILE,
    .reliability = UXR_RELIABILITY_RELIABLE,
    .history = UXR_HISTORY_KEEP_LAST,
    .depth = 10
};

uxrCustomTransport transport;
uxrSession session;
uxrStreamId reliable_out;
[... 89 lines cut ...]
    // Reliable QoS
    if(uxr_run_session_until_confirm_delivery(&session, 100))
        return 0;
    else
        return 1;
}

uint8_t create_publisher_imu()
{
    memcpy(msg_imu.header.frame_id,"base_link",9);
    msg_imu.orientation_covariance[0] = -1.0;
    msg_imu.orientation_covariance[4] = -1.0;
    msg_imu.orientation_covariance[8] = -1.0;
    return create_publisher("rt/imu/data", "sensor_msgs::dds_::Imu_");
}

```
# Trials

* use `uxr_buffer_create_xxxxxx_xml()` function instead but don't solve this problem.

# PS

* I've checked `rmw_xrcedds`, I copy the operation in it but still not working. I don't want to use it because it need too much ros2-related building environment tools like ament-cmake which seem weird in MCU project building.
* I've checked `PX4-Autopilot`, but it use template to generate related code and I don't learn to compile it.

pablogs9, 2023-12-21:
Your XRCE Client and XRCE Agent are working properly because the agent is publishing to DDS, as you can see in the log `[** <<DDS>> **]`. 

This is likely related to a:
1. QoS matching problem. What is the output of `ros2 topic info /your_topic -v`?
2. Serialization problem. Can you attach your type support?

Pansamic, 2023-12-21:
**1. QoS matching problem**

The output of `ros2 topic info /imu/data -v`(using Micro-XRCE-DDS):

```
❯ ros2 topic info /imu/data -v
Type: sensor_msgs/Imu

Publisher count: 1

Node name: nav_mecanum_driver
Node namespace: /
Topic type: sensor_msgs/Imu
Endpoint type: PUBLISHER
GID: 01.0f.88.f0.36.21.c5.a0.04.00.00.00.00.00.01.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 0

```

Output of `ros2 topic info /imu/data -v`(using Micro-ROS static library). This is the version that its message can be received by my ros2 nodes and can be echoed so it's my reference.
```
❯ ros2 topic info /imu/data -v
Type: sensor_msgs/msg/Imu

Publisher count: 1

Node name: nav_mecanum_driver
Node namespace: /
Topic type: sensor_msgs/msg/Imu
Endpoint type: PUBLISHER
GID: 01.0f.88.f0.36.21.c5.a0.03.00.00.00.00.00.0c.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 0
```

As you can see, `Topic type` is not the same. QoS is the same.

**2. Serialization problem**

I've noticed something about `rosidl_typesupport_microxrcedds` but I haven't use that. Is it essential? If so, how to use it? 
I find so much repos to find the dataType of Micro-XRCE-DDS of ROS2 message, like `PX4-Autopilot`, `ros2xrcedds`, `rmw_xrcedds` but I failed.

pablogs9, 2023-12-21:
The type name shall be fixed in order to have a proper DDS/ROS2 communication

Pansamic, 2023-12-21:
thanks, it works.

closed 2023-12-21 by Pansamic