eProsima/Micro-XRCE-DDS-Client #380: MicroRos fails in ra8m1 due to uxr_create_session

Imeguras, 2024-01-29:
i've tested code very similar(other than having to tweak low level ethernet, etc...)
Im using what is currently a non supported board/architecture however im really excited about the ra8 family and im trying to assemble a sort of tutorial for it
Im using the arm-gnu-toolchain-13.2.Rel1-x86_64-arm-none-eabi toolchain to be able to compile, etc... as llvm just gave me a nightmare and seems to break every so often(plus although i would consider it a better tool even on my linux machine, it gives me problems regardless of the type and scope of project)

Im using Azure RTOS, and to compile + link im using the latest version of GNU ARM

The Micro ROs breaks during the create session part, here(line 308 of rmw_init):

if (!uxr_create_session(&context_impl->session)) {
    CLOSE_TRANSPORT(&context_impl->transport);
    put_memory(&session_memory, &context_impl->mem);
    context->impl = NULL;
    RMW_UROS_TRACE_MESSAGE("failed to create node session on Micro ROS Agent.")
    return RMW_RET_ERROR;
}

But what really causes the issue is more precisely in(line 246 of session.c)

bool received = wait_session_status(session, create_session_buffer, ucdr_buffer_length(&ub), (size_t) retries);
    bool created = received && UXR_STATUS_OK == session->info.last_requested_status;

Where the received comes back as false breaking everything after it,
Although i've noticed in the past that due to not initializing variables it give the idea that they're glitched(since initialization only comes slightly before its used, giving a 10:1 floating/initialized perception), i've noticed that some kind of send on the session had a huge lenght and uxr_create_session(&context_impl->session)) seems to take like a few seconds(10 or so which is an eternity to what was in the ra6m5)

What can be causing this issue? I've tested the ethernet before and it seemed to work with some few tweaks(Addresses for the RMII interface, pins etc...)

I dont think there is any connection as to the micro agent on the pc as it reports the usual "blank" status
```md
[1706531270.750196] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1706531270.750664] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
```
(i've configured the micro ros to go to my domain_id(42))

pablogs9, 2024-01-29:
Your micro-ROS client is not connecting to the micro-ROS Agent, that's why `uxr_create_session` is failing. 

Did you check the connectivity between the MCU and the Agent?

Imeguras, 2024-01-29:
Actually it worked... there were just multiple problems related to both distraction and fatigue
Make shure to leave the J26 open(the tests were the network worked i had it open) but then with tiredness i might had it closed for some reason

Imeguras, 2024-01-29:
> Your micro-ROS client is not connecting to the micro-ROS Agent, that's why `uxr_create_session` is failing.
> 
> Did you check the connectivity between the MCU and the Agent?
Whoops sorry, i had no internet and i blindly coommented 
yes, that was the issue it was just me basically solving one issue while enabling another...

closed 2024-01-29 by Imeguras